sifive/freedom #1: Makefile.u500artydevkit

arunthomas, 2016-11-29:
Any chance you could commit a Makefile.u500artydevkit? I have an Arty board that I've been using as a U500. Thanks.

richardxia, 2016-11-29:
Hi Arun. We do plan to eventually release a version of the U500 Arty RTL, but we don't have a time frame for that yet. We can keep you posted when we have updates.

arunthomas, 2016-11-29:
Great. Thanks a lot, Richard.

terpstra, 2017-11-04:
Sorry that this issue has not gotten much attention. Unfortunately, I don't think this is going to happen any time soon. The U500 designs are rather too large to fit in the arty, especially now that a standard linux system expects an FPU. We may revisit this again in the future and/or use a newer arty-like board that has more LUTs. Sorry.

closed 2017-11-04 by terpstra