sifive/freedom #2: Make verilog fails on OSX

impedimentToProgress, 2016-11-29:
sed -r 's/\(([A-Za-z0-9_]+),([A-Za-z0-9_]+)\)/`define \1 \2/' /Users/maXXXX/Desktop/sifive/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.prm >> /Users/maXXXX/Desktop/sifive/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.vh
sed: illegal option -- r
usage: sed script [-Ealn] [-i extension] [file ...]
       sed [-Ealn] [-i extension] [-e script] ... [-f script_file] ... [file ...]
make: *** [/Users/maXXXX/Desktop/sifive/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.vh] Error 1

wumpus, 2016-11-29:
http://stackoverflow.com/questions/30003570/how-to-use-gnu-sed-on-mac-os-x

richardxia, 2016-12-01:
Thanks for reporting this. While we probably won't guarantee 100% support for OS X (most FPGA synthesis tools don't support it), I looked into this particular case and it turns out that the `.prm` file referenced in the `sed` command is deprecated in `rocket-chip` anyway, so I was able to remove it altogether.

PR #6 (2016-12-01): Remove verilog header files built from Chisel .prm file.
Will coincidentally fix #2.

The `.prm` file that rocket-chip's Generator outputs is now deprecated and emits an empty file, so now there's no point in creating the `.vh` file from it. The coincidentally caused an incompatibility with BSD-based systems like macOS, since we were using a GNU-only `sed` flag to generate the `.vh` file.

This PR removes the `.vh` file and all references to it from the FPGA scripts. As a sanity check, I've confirmed that the E300 Arty Dev Kit's MCS file still works on an actual Arty board. I haven't tested th [... cut]
closed 2016-12-01 by richardxia