sifive/freedom #5: Is it possible to boot Linux?

MarcoGeuna, 2016-11-30:
Hi all,

Is it possible to boot Linux from SD card on your system (i.e. VC707 FPGA Dev Kit)?
I'm searching something like the Rocket-chip untethered implementation made by LowRISC...

Thank you!

terpstra, 2016-11-30:
Yes. Run make in the freedom-u-sdk repository. Put SD card into vc707 fpga.

MarcoGeuna, 2016-12-01:
Perfect, thank you @terpstra !
But, what about if I modify the Chisel code provided by you...will I be able to boot Linux the same?
I mean, my idea is to remove the "HiTechGlobal X8 PCI Express Gen1/2/3 FMC Module", which you list as required hardware in the [SiFive Freedom U500 VC707 FPGA Getting Started Guide](https://dev.sifive.com/develop/freedom-u500-vc707-fpga-dev-kit/). I want to have the board as a "standalone component" (connected with a computer via UART/Ethernet if needed). I can modify the RTL code, but I prefer not to change the software components...hence basically what you provide inside the freedom-u-sdk repository.

Thank you again!

terpstra, 2016-12-01:
PCIe is not required to boot linux. However, you will need to modify the design a bit because we receive the system clock from that FMC. Also, we do not have RTL to use the Ethernet on the VC707. The network card we were using was PCIe.

MarcoGeuna, 2016-12-06:
Got it! Thank you so much for your answer! I'll try to do that

MarcoGeuna, 2017-01-18:
Hi again!
I probably succeeded in removing the PCIe dependency (actually it's still present but I derive the main clock from another component I added). To boot Linux on my "Freedom U500" I also followed what @terpstra told me 

> Run make in the freedom-u-sdk repository. Put SD card into vc707 fpga. 

Unluckily, something doesn't work properly, cause the booting sequence I see using Minicom is only the following:
" INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT "

After that it stops and doesn't continue like you show in your _"SiFive Freedom U500 VC707 FPGA Getting Started Guide"_.

In this moment I'm not thinking about an HW design problem: the clock should be the right one (62.5 MHz, as I asked in other issues ([13](https://github.com/sifive/freedom/issues/13) and [12](https://github.com/sifive/freedom/issues/12))), and, taking into account that the system starts, I think I am also giving the right reset. Hence, I fear the problem is a SW one.
I'm not a SW expert, so I literally only do what @terpstra  told me. But maybe I am missing something...one hypothesis (but I don't know how to prove it!) Is that my system starts with a sort of first stage bootoader, that later fails looking for the actual kernel image.

Any hint, suggestion or idea will be very useful! Thank you so much!

terpstra, 2017-01-19:
What version of rocket-chip are you using? There was a recent change on master that broke SMP boot.

MarcoGeuna, 2017-01-19:
Well, actually I don't know how to find the rocket-chip version I am using! Sorry!
But, with the command _**git log**_, I can tell you the latest commit I have on my local machine:

- on _**freedom**_ repo:  commit f4375c22662f82b1b4f94e88b1aba6998b1f34ba (Thu Dec 8 2016)
- on _**rocket-chip**_ repo (inside _freedom_ one): 4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7 (Sat Nov 26 2016)
- on _**freedom-u-sdk**_ repo: commit b38f7c98485e78b9638a04e499d01fcb8ebf2958 (Fri Dec 2 2016)

Is it enough for you to understand my rocket-chip version? Otherwise, please, tell me where to see it.
Thank you!

terpstra, 2017-01-19:
That should have worked. So I would say that your problem is indeed software. What command did you use to write the image to the SD card?

MarcoGeuna, 2017-01-19:
What I do is, in order:

1. `git clone --recursive https://github.com/sifive/freedom-u-sdk.git` to clone _freedom-u-sdk_ repo
2. `make` inside the repo, to create the bbl.bin
3. `sudo dd if=work/bbl.bin of=/dev/sdc bs=1M` to copy it to the SD card

Anything wrong?

I add this: I didn't reconfigure both the _linux kernel_ and  _buildroot_. I don't know much about them, so I started supposing that, for me, the basic "make bbl.bin" was sufficient. Maybe this wasn't right...

terpstra, 2017-01-20:
Ok. That looks fine. I think you're going to need to dive into waveforms to debug this... The first thing the boot loader does is copy some stuff around. Is your DDR region at least as large as what the config string claims?

MarcoGeuna, 2017-01-20:
When you say 

> Is your DDR region at least as large as what the config string claims?

I think you are referring to the **_ConfigStringOutput.contents_** inside _freedom/src/main/scala/unleashed/u500vc707devkit/Top.scala_. Am I correct? If yes, that's what I have inside (I didn't modify this file):

|ram {
|    0 {
|      addr 0x80000000;
|      size 0x10000000;
|     };
|};

If I understand correctly, this gives 256 MB to RAM. And on my VC707 I have 1 GB. Hence I'd answer yes, to your question: my DDR region should be large enough.

MarcoGeuna, 2017-01-20:
Another idea for this non working system could be this. Let me describe my situation:

1. I derived the `top_clock` from a PLL I added. This because I don't have the HiTechGlobal X8 PCI Express Gen1/2/3 FMC Module
2. despite this, the "HiTechGlobal X8 PCI Express..." is still present in my design, at the level of Chisel description (I'm bad with Chisel: when I remove all the `xilinxvc707pcie` stuff **from Top.scala** I get an error with `l2.TLAdapterNode` I'm not able to solve)
3. as a consequence, I only removed the `xilinxvc707pcie` connections related to clock and reset, leaving all the others. Then, I add my module at the level of verilog code, after giving "make verilog"

I wonder if some of the connections I left are blocking the system. For example, maybe the Coreplex at a certain point wants an input from the PCIe...but, since the PCIe is not physically present, this will not arrive, causing the system to stall. Is this possible?

terpstra, 2017-01-23:
I see that you managed to removed PCIe now. However, I doubt this will solve your problem; PCIe is enabled well after linux has started to boot and you'd have seen output. Ultimately, I am pretty sure you're going to need to attach chipscope to debug the problem. Here's an example of a script I personally apply to the verilog to debug it:

```
awk '
/^module TLXbar_l1tol2/ { f=1 };
/^);/ { f=0 };
/io_in/ && f { printf "(* mark_debug=\"true\" *)" };
/io_in/ && f && !/data/ && !/mask/ && !/addr/ { printf "(* mark_debug_trigger=\"true\" *)" };
1'
```

I run this manually on the verilog after firrrtl runs and before Vivado is started. Then after you program the FPGA, you can attach to chipscope using the generated '*_early.ltx' file. That lets your monitor all the TileLink (bus) traffic in the system. Usually that gives a good indication of where it hangs.

terpstra, 2017-01-23:
Shoot. I don't think this script will work for you. It relies on some parts of our company internal flow. Anyway, you get the point: you want to run chipscope to figure out at what point the boot has stopped.

MarcoGeuna, 2017-01-24:
Thank you @terpstra for your answer, I appreciated! I already saw your suggestion right after you wrote it, but, to reply, I was waiting to have some results. Unluckily now I have to move to other tasks related to U500VC707 FPGA DevKit ([like this](https://github.com/sifive/freedom/issues/16)). So, if I will have to come back, trying to boot Linux, I will follow what you told me!

But, before that, one last question. When you wrote me this:
> I see that you managed to removed PCIe now

I suppose you were thinking about what I described [in this other issue](https://github.com/sifive/freedom/issues/14).  Well, related to that issue, I fear that, although I am able to generate verilog without errors, my work is incomplete. **I only removed**:

1. all the _"with PeripheryXilinxVC707PCIeX1 * "_ for the various _U500VC707DevKitSystem *_ classes (inside Top.scala)
2. all the connections  _XilinxVC707PCIeX1.**master **_   has with **_l1tol2 TLXbar_**, among different traits and classes (in files: _freedom/src/main/scala/coreplex/Coreplex.scala_ and _freedom/src/main/scala/rocketchip/RocketPlexMaster.scala_).

With this, the PCIe disappears from _Interrupr Map_, _Generated Address Map_ and _Generated Configuration String_. I also tried to remove the **_Xilinx AXI PCIe host bridge support fo RV_** from the linux kernel, using command _make linux-menuconfig_ inside _freedom-u-sdk_ repo. The result is always the same, and I'm not able to get over the BOOT message.

Btw, **I didn't change anything connected to the following XilinxVC707PCIeX1 val**: (inside _/freedom/sifive-blocks/src/main/scala/devices/xilinxvc707pciex1/XilinxVC707PCIeX1.scala_)

- **slave** [TLInputNode()]
- **control** [TLInputNode()]
- **intnode** [IntSourceNode(1)]

I dont' know this PCIe IP from Xilinx, nor how it interacts with your SoC. Ok, Chisel doesn't give any error, but maybe who knows, since it should be connected to the interrupt system (thorough _intnode_), I encounter some problem. Can my "partial removing" lead me to deadlocks?

Thank you so much!

closed 2017-05-19 by richardxia

hylee17, 2017-09-27:
@MarcoGeuna 
Hi. I'm also trying to boot linux on VC707 without an FMC Card..
How did you fix the system clock?.?