sifive/freedom #9: git clone --recursive fails

impedimentToProgress, 2016-12-12:
It worked a couple of weeks ago, but now I (and a colleague) get this error message:

Cloning into 'env'...
fatal: read error: Operation timed out
fatal: clone of 'git://github.com/ucb-bar/riscv-test-env.git' into submodule path 'env' failed

richardxia, 2016-12-12:
I'm able to recursively clone myself, but I am wondering if you and your colleague are trying to clone in an environment that has blocked the port used for the `git://` protocol ([apparently 9418](https://git-scm.com/book/ch4-1.html#The-Git-Protocol)).

The actual submodule reference is two levels deep, living in `rocket-chip` -> `torture` -> `env`: https://github.com/ucb-bar/riscv-torture/blob/77195ab12aefc373ca688e0a9c4d710c13191341/.gitmodules#L3. I can submit a patch to change it to `https://` to make it consistent with all the other submodule URLs, but it may take some time to pull the changes through all the submodules to this repo and verifying that the other changes pulled this way don't break any functionality.

In the meantime, if you are unable to get that port opened up, you can manually change the URL to that submodule by `cd`ing into the `rocket-chip/torture/env` directory and running:

```sh
$ git remote set-url origin https://github.com/riscv/riscv-test-env.git
```

closed 2016-12-12 by richardxia

impedimentToProgress, 2016-12-12:
The port was open (because it was working before and the other submodules got downloaded).  I thing going to https is a much better method in general as it doesn't require me to create and add a public key to github.

sorear, 2016-12-12:
ssh (`git@github.com:riscvriscv-test-env`) requires a public key, git (`git://github.com/...`) and https don't