sifive/freedom #10: Modify the internal Rocket parallelism

MarcoGeuna, 2016-12-14:
Hi all,

is it possible to modify the internal CPU parallelism? (i.e. the Rocket Core one) I looked at the Chisel code, but I'm not sure if I understood right.
Is it possible to switch between a 32 bits and a 64 bits (and 128 bits) CPU? In other words: is the Rocket-Chip architecture able to implement all the RV32, RV64 (and RV128) RISC-V ISAs?

If yes, is there a configuration parameter able to do that "automatically"? Or do I have to modify something by hand?

Thank you so much!

aswaterman, 2016-12-14:
The XLen parameter is what you're looking for, but only 32 and 64 are supported.  (We don't support RV128 yet because there's no software support for it, so it isn't readily testable.)

MarcoGeuna, 2016-12-14:
Perfect! This means I understood right! Thank you very much @aswaterman for your fast answer!

closed 2017-05-19 by richardxia