sifive/freedom #11: How to simulate through the VLSI tools

colin4124, 2016-12-26:
I am studying debug mechanism (JTAG with OpenOCD). I reference https://github.com/riscv/riscv-tests/tree/master/debug to figure out that. I haven't SiFive Board (e300, u500), so I want to try in Simulation (`./gdbserver.py --freedom-e300-sim`)

I notice that reedom/src/main/scala/everywhere/e300artydevkit/Configs.scala, maybe I could do something like that:
```
$ cd freedom/rocket-chip/vsim/
$ make verilog CONFIG=E300ArtyDevKitConfig
$ cd riscv-tests/debug/
$ ./gdbserver.py --freedom-e300-sim --run simv-rocketchip--E300ArtyDevKitConfig
```

So, what I should do to achieve that?

richardxia, 2017-01-03:
I'm afraid I don't understand exactly what you are asking. I assume you are trying to simulate using VCS. What problems are you running into and how are you building the simulator?

colin4124, 2017-01-04:
I am trying to simulate freedom-e300 using VCS. And then, I want to use GDB to debug it.

I found that [riscv-tests](https://github.com/riscv/riscv-tests/tree/master/debug) have already written a python program to test Debug interface of freedom-e300.  But [that](https://github.com/riscv/riscv-tests/tree/master/debug#32-bit-rocket-chip-core-in-simulation) is quite simple:
`./gdbserver.py --freedom-e300-sim`

I think just using this command would get an error, because I have no VCS simulator of freedom-e300 yet. So I have to generate a  VCS simulator of freedom-e300 like [that](https://github.com/ucb-bar/rocket-chip#building-the-project) :
```
$ cd freedom/rocket-chip/vsim
$ make CONFIG=E300ArtyDevKitConfig
```
But it doesn't work, it can't recognize the **E300ArtyDevKitConfig** configuration. Because `Makefile` in freedom/rocket-chip/vsim set the current project to rocket-chip not the upper level freedom, and the **E300ArtyDevKitConfig** configuration is defined in freedom level.

After that, I found [project-template](https://github.com/ucb-bar/project-template). It has **vsim Folder** but freedom has not. So I just copy to freedom, and then edit Makefile:
```
PROJECT ?= example
MODEL ?= TestHarness
CONFIG ?= DefaultExampleConfig
CFG_PROJECT ?= $(PROJECT)
TB ?= TestDriver
```
to
```
PROJECT ?= sifive.freedom.everywhere.e300artydevkit
MODEL ?= E300ArtyDevKitTop
CONFIG ?= E300ArtyDevKitConfig
CFG_PROJECT ?= $(PROJECT)
TB ?= TestDriver
```
But it does not work. Maybe I have to edit them too:
```
sim_vsrcs = \
	$(build_dir)/$(PROJECT).$(CONFIG).v \
	$(build_dir)/$(PROJECT).$(CONFIG).behav_srams.v \
	$(base_dir)/rocket-chip/vsrc/TestDriver.v \
	$(base_dir)/testchipip/vsrc/SimSerial.v

sim_csrcs = \
	$(base_dir)/testchipip/csrc/SimSerial.cc

VCS = vcs -full64

VCS_OPTS = -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet \
	+rad +v2k +vcs+lic+wait \
	+vc+list -CC "-I$(VCS_HOME)/include" \
	-CC "-I$(RISCV)/include" \
	-CC "-std=c++11" \
	-CC "-Wl,-rpath,$(RISCV)/lib" \
	$(RISCV)/lib/libfesvr.so \
	-sverilog \
	+incdir+$(generated_dir) \
	+define+CLOCK_PERIOD=1.0 $(sim_vsrcs) $(sim_csrcs) \
	+define+PRINTF_COND=$(TB).printf_cond \
	+define+STOP_COND=!$(TB).reset \
	+define+RANDOMIZE_MEM_INIT \
	+define+RANDOMIZE_REG_INIT \
	+define+RANDOMIZE_GARBAGE_ASSIGN \
	+define+RANDOMIZE_INVALID_ASSIGN \
	+libext+.v \
```
But I don't know what to edit? I can't building the simulator of freedom-e300

richardxia, 2017-01-04:
Just to be clear, it is outside the scope of rocket-chip to provide a factory for building RTL simulators. However, I can attempt to help you with what you're trying to do. Just bear in mind that this is not a supported workflow, and it could break in future versions of rocket-chip or freedom.

I'm not familiar with `project-template`, and I do not know if it is being kept up to date with the latest rocket-chip, so it may be easier to manually generate the verilog files for the E300ArtyDevKit using the `Makefile.e300artydevkit` that comes with this freedom repo.

In addition, you will need to create your own TestHarness in order to instantiate the JTAGVPI module and connect it to the design under test. Without one, you will not actually have a debugging module in the generated verilog, and you'll be unable to actually talk to the debugger with OpenOCD.

You may be able to copy the rocketchip `TestHarness.scala` and make the changes that I described in https://github.com/riscv/riscv-tests/issues/39#issuecomment-269756720. Note that you should also change the `dut` to instantiate your design and not the ExampleRocketTop. You will then need to change `Makefile.e300artydevkit` so that `MODEL` is set to the name of your TestHarness module.

Once you generate those, you may be able to modify the `sim_vsrcs` list in the `vsim/Makefile` to include the Chisel-generated verilog source file along with all the behavioral verilog and other black box models. You should be able to drop all the existing files from `sim_vsrcs` and just include the manually generated verilog file, the extra verilog files listed in `EXTRA_FPGA_VSRCS` from `Makefile.e300artydevkit`, and rocket-chip's `vsrc/jtag_vpi.v`. For `sim_csrcs`, drop the `SimSerial.cc` and add rocket-chip's `csrc/jtag_vpi.c`.

To summarize, here are the high level changes you need to make:

- Create your own TestHarness that wraps and connects the E300ArtyDevKit and JTAGVPI modules
- Build the design verilog from Chisel
- Include the design verilog as well as all behavioral and other blackbox verilog files when building the VCS simulator

closed 2017-05-19 by richardxia