sifive/freedom #12: UART matters

MarcoGeuna, 2017-01-11:
Hi all,

I'd like to ask something about the UART in the U500 platform. In particular:
1)  At which clock does it operate? Think about the process inside file:
                 freedom/fpga/u500vc707devkit/src/system.v
     It is clocked using signal top_clock. Which is its frequency?

2) How do we have to set Minicom, to communicate with the vc707 FPGA?
    In your "SiFive-U500-vc707-gettingstarted-v0.2" guide, there are many parameters, like "Stop bits = 1".
     But if I look inside file:
                   freedom/sifive-blocks/src/main/scala/devices/uart/UART.scala 
     I see: ``stopBits: Int = 2,``
     So, I am a bit confused! Which are the actual parameters?

Thank you for everything!

terpstra, 2017-01-12:
Re #1: 62.5MHz, although other frequencies can work because the UART uses a clock divider to reach 115200Hz for signaling.

Re #2: My guess is that value is the maximum supported stopBits and you control the actual value at runtime by a register. However, I've asked albert to comment; he knows for certain.

a0u, 2017-01-12:
> My guess is that value is the maximum supported stopBits and you control the actual value at runtime by a register.

This is correct.  The reset value of the `txctrl` register is given [here](https://github.com/sifive/sifive-blocks/blob/7916ef5249c72a3a84c599d123760f4d716de58a/src/main/scala/devices/uart/UARTCtrlRegs.scala#L7), which corresponds to one stop bit.

MarcoGeuna, 2017-01-18:
Thank you to both for this explanation!

closed 2017-01-18 by terpstra