sifive/freedom #14: Unknown error doing "make verilog"

MarcoGeuna, 2017-01-22:
Hi all! I'm trying to modify the **_u500vc707devki_** project, removing the XilinxPCieX1 component. I worked on the Chisel code: in a few words, from Top.scala I removed all the stuff related to XilinxPCieX1. Then, I modified **_trait RocketPlexMaster_**, making it **_extends TopNetwork_** instead of **_L2Crossbar_**.

When I give command `make -f Makefile.u500vc707devkit verilog`, the verilog is produced, but I obtain an error. As you can see from the log (reported below), it doesn't say anything meaningful about what happened (or maybe I'm not able to see it!). Only an "**Error 1**" at the end.

**My question is:** is there a way to understand what happened? Maybe something like more output on screen, using an option via command line.

Thanks!

----------------------------------------------------------------------------------------------------------------------------------------------
**ERROR LOG:**

mkdir -p /home/marco/repo/freedom/builds/u500vc707devkit/
java -jar /home/marco/repo/freedom/rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /home/marco/repo/freedom/builds/u500vc707devkit sifive.freedom.unleashed.u500vc707devkit U500VC707DevKitTop sifive.freedom.unleashed.u500vc707devkit U500VC707DevKitConfig"
[info] Loading project definition from /home/marco/repo/freedom/rocket-chip/project
Using addons: 
[info] Set current project to freedom (in build file:/home/marco/repo/freedom/)
[info] Compiling 1 Scala source to /home/marco/repo/freedom/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Compiling 1 Scala source to /home/marco/repo/freedom/target/scala-2.11/classes...
[info] Running rocketchip.Generator /home/marco/repo/freedom/builds/u500vc707devkit sifive.freedom.unleashed.u500vc707devkit U500VC707DevKitTop sifive.freedom.unleashed.u500vc707devkit U500VC707DevKitConfig
[info] [0.002] Elaborating design...

Interrupt map:
  [1, 1] => uart0
  [2, 2] => spi0
  [3, 6] => gpio

Generated Address Map
	xilinxvc707mig 80000000 - 90000000, RWX [C]
	uart0 54000000 - 54001000, RW
	spi0 54001000 - 54002000, RW
	bootrom 1000 - 2000, RX
	debug 0 - 1000, RWX
	gpio 54002000 - 54003000, RW
	clint 2000000 - 2010000, RW
	plic c000000 - 10000000, RW

Generated Configuration String
platform {
  vendor ucb;
  arch spike;
};
plic {
  interface "plic";
  ndevs 9;
  priority { mem { 0x0c000000 0x0c00ffff; }; };
  pending  { mem { 0x0c001000 0x0c00107f; }; };
  0 {
    0 {
      m {
        ie  { mem { 0x0c002000 0x0c00207f; }; };
        ctl { mem { 0x0c200000 0x0c200007; }; };
      };
      s {
        ie  { mem { 0x0c002080 0x0c0020ff; }; };
        ctl { mem { 0x0c201000 0x0c201007; }; };
      };
  };
  };
};
leds {
  interface "gpio";
  ngpio 4;
  mem { 0x54002000 0x54002003; };
};
rtc {
  addr 0x200bff8;
};
ram {
  0 {
    addr 0x80000000;
    size 0x10000000;
  };
};
uart {
  addr 0x54000000;
};
core {
  0 {
    0 {
      isa rv64ima;
      timecmp 0x02004000;
      ipi 0x02000000;
    };
  };
};

[info] [6.936] Done elaborating.
[success] Total time: 21 s, completed Jan 21, 2017 7:58:29 PM
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/marco/repo/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.fir -o /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v -X verilog
sed -i -s "s/ *output\(.*\)__inout\(.*\)/inout \1__inout\2/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_xilinxvc707mig__inout_ddr3_dq(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dq)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_xilinxvc707mig__inout_ddr3_dq(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dq),/    .io_xilinxvc707mig__inout_ddr3_dq(io_xilinxvc707mig__inout_ddr3_dq),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_xilinxvc707mig__inout_ddr3_dqs_n(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dqs_n)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_xilinxvc707mig__inout_ddr3_dqs_n(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dqs_n),/    .io_xilinxvc707mig__inout_ddr3_dqs_n(io_xilinxvc707mig__inout_ddr3_dqs_n),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_xilinxvc707mig__inout_ddr3_dqs_p(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dqs_p)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_xilinxvc707mig__inout_ddr3_dqs_p(U500VC707DevKitSystem_1_io_xilinxvc707mig__inout_ddr3_dqs_p),/    .io_xilinxvc707mig__inout_ddr3_dqs_p(io_xilinxvc707mig__inout_ddr3_dqs_p),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_port__inout_ddr3_dq(xilinxvc707mig_io_port__inout_ddr3_dq)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_port__inout_ddr3_dq(xilinxvc707mig_io_port__inout_ddr3_dq),/    .io_port__inout_ddr3_dq(io_xilinxvc707mig__inout_ddr3_dq),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_port__inout_ddr3_dqs_n(xilinxvc707mig_io_port__inout_ddr3_dqs_n)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_port__inout_ddr3_dqs_n(xilinxvc707mig_io_port__inout_ddr3_dqs_n),/    .io_port__inout_ddr3_dqs_n(io_xilinxvc707mig__inout_ddr3_dqs_n),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "    .io_port__inout_ddr3_dqs_p(xilinxvc707mig_io_port__inout_ddr3_dqs_p)" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/    .io_port__inout_ddr3_dqs_p(xilinxvc707mig_io_port__inout_ddr3_dqs_p)/    .io_port__inout_ddr3_dqs_p(io_xilinxvc707mig__inout_ddr3_dqs_p)/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "   .ddr3_dq(blackbox_ddr3_dq)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/   .ddr3_dq(blackbox_ddr3_dq),/   .ddr3_dq(io_port__inout_ddr3_dq),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "   .ddr3_dqs_n(blackbox_ddr3_dqs_n)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/   .ddr3_dqs_n(blackbox_ddr3_dqs_n),/   .ddr3_dqs_n(io_port__inout_ddr3_dqs_n),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && grep -q -F   "   .ddr3_dqs_p(blackbox_ddr3_dqs_p)," /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v && sed -i -s  "s/   .ddr3_dqs_p(blackbox_ddr3_dqs_p),/   .ddr3_dqs_p(io_port__inout_ddr3_dqs_p),/g" /home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v 
make: *** [/home/marco/repo/freedom/builds/u500vc707devkit/sifive.freedom.unleashed.u500vc707devkit.U500VC707DevKitConfig.v] Error 1

richardxia, 2017-01-22:
Hi @MarcoGeuna, the command that failed is actually displayed in the output you see, but it's a bit difficult to recognize because it's actually a 20-line chain of `sed` commands that correspond to this part of the Makefile: https://github.com/sifive/freedom/blob/f4375c22662f82b1b4f94e88b1aba6998b1f34ba/Makefile.u500vc707devkit#L20-L25

The current version of Chisel does not support bidirectional pins (i.e. `inouts`), so this part of the Makefile contains a post-processing step to transform wires that contain `_inout` in their name into `inouts`. Since you are removing the Xilinx PCIe device, you should also remove the corresponding lines that I linked to above.

MarcoGeuna, 2017-01-22:
Ok, nice! That worked, removing the error!

Sorry @richardxia, probably this was not so difficult to understand by myself...next time I'll be more careful! Thank you so much!

richardxia, 2017-01-23:
You're welcome, and don't be afraid to ask future questions!

closed 2017-01-23 by richardxia

MarcoGeuna, 2017-01-24:
Sorry @richardxia, I have another question for you, related to this topic.
Ok, as I told you, now this problem is solved; but the system is still not doing what I want. For this reason I am trying to find the error, and I had a question, related to your answer:

> Since you are removing the Xilinx PCIe device, you should also remove the corresponding lines that I linked to above.

The question is: since I am trying to remove the Xilinx _**PCIe**_ device, why did you tell me to remove lines related to the Xilinx **_MIG_** device? I looked inside the verilog code (in both cases: with and without PCIe) and MIG and PCIe seems to be disconnected...so why "removing" PCIe gives me error on MIG outputs? (ok, MIG _inouts_ ;-) )

richardxia, 2017-01-24:
Ah, sorry, I didn't pay close enough attention to that. I reproduced your error locally and noticed that one of the `grep`/`sed` commands failed because it was looking for a comma at the end of the line, but in the verilog that port is now the last item in the list so there is no longer a trailing comma. Remove all the commas in https://github.com/sifive/freedom/blob/f4375c22662f82b1b4f94e88b1aba6998b1f34ba/Makefile.u500vc707devkit#L24-L25 (lines 24 and 25), and it should succeed.

MarcoGeuna, 2017-01-24:
Yes, it does! It doesn't give the problem anymore, and now this solution makes sense!
Thank you again!