sifive/freedom #15: Config of u500vc707devkit

MarcoGeuna, 2017-01-22:
Hi all,

studying your repo, I understood that probably in your system (**given by** the configuration inside **_U500VC707DevKitConfig_**) you:

- do **not** use L2 cache (I didn't find as included any of the **_WithL2Cache_** configs or similar)
- do **not** use FPU (you include configs **_WithoutFPU_**; then I also had a look at how its parameters as used by tile.scala and rocket.scala)

Am I correct? If not, can you please tell me where you add those components?

Thank you all!

terpstra, 2017-01-23:
You are correct. The FPGA development kit does not at present enable these options.

MarcoGeuna, 2017-01-23:
Perfect! Thank you so much @terpstra !

closed 2017-05-19 by richardxia