sifive/freedom #16: Bare metal tests on U500 VC707 FPGA Dev Kit

MarcoGeuna, 2017-01-24:
Hi again!
I wonder if it is possible to run simple bare metal test on the system you provide. I think so (it is able to boot Linux!), hence I'm trying to find a way.

I find myself a bit confused, since actually there is a lot of material, like inside [riscv-tools](https://github.com/riscv/riscv-tools/tree/cd78e37f72cfc2a452a0c11744586084fbae1dcd) repo. For example [riscv-tests](https://github.com/riscv/riscv-tests/tree/6a1a38d421fd3e24bdc179d58d33572636b903b2), or also [riscv-benchmarks](https://github.com/ucb-bar/riscv-benchmarks). Despite this, it is not clear to me if they are related to the new untethered Rocket Chip, or to the old tethered one. Sometimes, since their last update is prior to the untether, I fear they are not good for me.

Can please tell me how to do bare metal tests? Or, if you prefer, pointing me to a repo or somewhere I can understand by myself how to do them?

Thank you so much again!

richardxia, 2017-01-24:
The tests in `riscv-tests` are written in a way that's agnostic to whether or not the chip is tethered. You will need a way of loading the tests onto the chip. One way of doing so for the U500 VC707 Dev Kit is to put the compiled test on an SD card and load it similar to how Linux can be booted.

MarcoGeuna, 2017-01-24:
Perfect! I'll try to do that...thank you so much again!

MarcoGeuna, 2017-01-30:
Hi, I'm here again... I had a look at the `riscv-tests` repo you pointed me. I was able to run all the tests in `riscv-tests/benchmarks`, using `spike pk <test_name>`.

But, since my aim is to run them on FPGA as bare metal, I was looking also to some files like drivers of common peripherals, like the uart. I found some files with name ending with *uart* inside `riscv-gnu-toolchain` repo, but, before starting long and maybe a priori wrong tests, I prefer to ask you directly.

I tried to run on FPGA the _median_main_ benckmarch inside `riscv-tests/benchmarks`, but it didn't work (i.e. no output on uart). This may be due to one error from me, or to the lack of drivers (e.g. uart ones).

For this reason I want to ask you: are bare metal drivers included with the compiler, or where are them? And, if present, how can I use them (e.g. link them to an executable file)?

Thank you again!

richardxia, 2017-01-30:
The closest thing we currently have to drivers are https://github.com/sifive/freedom-e-sdk/tree/master/bsp/libwrap. You can specifically take a look at https://github.com/sifive/freedom-e-sdk/blob/cebc4033491f5be7b717ad2ac66715771d374138/bsp/libwrap/sys/write.c, which replaces the `write` UNIX syscall (which doesn't exist bare metal) with something that will write to the UART. Before you can call that code, you will need to initialize the UART with this code: https://github.com/sifive/freedom-e-sdk/blob/3d19e53e38d0e3f3785f159dd08668ab50e11e5a/bsp/env/freedom-e300-arty/init.c#L36-L42

Note that all the code I've linked to above was written for the E300 platform, so you may need to adjust a few things such as the address maps to make sure they work on the U500.

MarcoGeuna, 2017-01-31:
Ok, thank you! I will try this!

closed 2017-05-19 by richardxia