sifive/freedom #18: DRAM connection inside U500 VC707 FPGA Dev Kit

MarcoGeuna, 2017-01-30:
Hi all,

I have some troubles understanding how the the DRAM is connected with the rest of the system inside U500 VC707 FPGA Dev Kit. Here below there is what I understood (with some guesses).

On VC707 board, DRAM are connected on an AXI bus. MIG is the memory controller, so the one in charge of interacting with DRAM. But how MIG is connected to the rest of the SoC? Through TileLink.

Looking inside Chisel code, and the verilog output, I was able to reconstruct this: 
1) let's start from file:
freedom/sifive-blocks/src/main/scala/devices/xilinxvc707mig/XilinxVC707MIG.scala
_**class XilinxVC707MIG**_: here we find this connections: axi4 -> toaxi4.node -> xing.node -> node

2) file:
 /freedom/sifive-blocks/src/main/scala/devices/xilinxvc707mig/XilinxVC707MIGPeriphery.scala
the component XilinxVC707MIG described before is connected thorough
`val mem = Seq(xilinxvc707mig.node)`

3) this `val mem`, (**I GUESS**) is defined (and connected to coreplex) inside file:
freedom/rocket-chip/src/main/scala/rocketchip/RocketPlexMaster.scala

I don't want to annoy you too much guys, so please only tell me if what I wrote is correct, and if not, try to give me an idea of how things are organized.

Sorry for this question, and thank you for the answer!

terpstra, 2017-01-30:
That is correct.

MarcoGeuna, 2017-01-31:
Perfect! Thank you so much @terpstra!

closed 2017-05-19 by richardxia