sifive/freedom #19: Features of TileLink2

MarcoGeuna, 2017-01-31:
Hi all,

I noticed that, apparently, moving from TileLink "made" with package **_tilelink_** and then with package **_tilelink2_**, you also changed the implementation!

A part from adding stuff like _TLWidthWidget_, _TLAtomicAutomata_ etc, if now I have a look at TileLink buses inside the verilog output from your design in this repo, I don't see any of the stuff described in [the overview of the TileLink Protocol](https://docs.google.com/document/d/1Iczcjigc-LUi8QmDPwnAu1kH4Rrt6Kqi1_EUaCrfrk8/pub).

I saw that now there are other "channels", called **a**, **b**, **c**, **d** and **e**; I suppose they replace the old Acquire, Probe, Release, Grant and Finish channels. A part from that, also the signals they use are different from the one of the [overview](https://docs.google.com/document/d/1Iczcjigc-LUi8QmDPwnAu1kH4Rrt6Kqi1_EUaCrfrk8/pub).

What I want to ask you is this: is available some documentation about TileLink2? If not, how much does it change from "TileLink1", in the protocol, the channel aims...

Everything can be useful! Thank you so much!

terpstra, 2017-01-31:
We will be releasing the TL2 specification soon. It is slightly out-of-date, so we don't want to post it just yet.

MarcoGeuna, 2017-01-31:
OK, that's fair. Thank you for answering!

Do you have, approximately, an idea about the release period for this new specification?

terpstra, 2017-01-31:
@hcook

andrewhlin, 2017-02-11:
Also hoping to see a spec soon!

xfguo, 2017-03-02:
TL2 may want to design a flexible, scalable interconnect framework that can replace all the AMBA (axi,ahb,apb) interconnect (from hi-speed to low-speed), and use converter (to ahb/axi/apb) to connect the traditional IP cores w/ AMBA interface? Am i right? @terpstra

terpstra, 2017-03-02:
That's exactly how we've been using it, yes. Inside the SoC it's all cache coherent TL2. When we need to talk to third-party IP, we go through a converter to an AMBA protocol.

xfguo, 2017-03-02:
@terpstra Thanks for you reply.

But what if I don't need to use the *cache coherent* feature, Just like we want to use **rocket** to build a MCU, it's single core, like... STM32 or NXP LPC series.

terpstra, 2017-03-02:
Sure... when cache coherence is unneeded, the TL2 components do not instantiate the coherency hardware. So it can be used analogously to AXI.

mwachs5, 2017-08-22:
The Tile Link 2 Spec is now available on SiFive's documentation website:
https://www.sifive.com/documentation/tilelink/tilelink-spec/

closed 2017-08-22 by mwachs5

xfguo, 2017-08-23:
@mwachs5 That's Great!