sifive/freedom #20: mcs build problem

esden, 2017-03-12:
Hi I am trying to synthesize the arty bitstream but I am getting an error. Any suggestions?

```
% make -f Makefile.e300artydevkit mcs    
VSRC_TOP=/media/esdentem/Venture_Data/projects/riscv/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v EXTRA_VSRCS="/media/esdentem/Venture_Data/projects/riscv/freedom/rocket-chip/vsrc/AsyncResetReg.v /media/esdentem/Venture_Data/projects/riscv/freedom/rocket-chip/vsrc/DebugTransportModuleJtag.v /media/esdentem/Venture_Data/projects/riscv/freedom/sifive-blocks/vsrc/SRLatch.v" make -C /media/esdentem/Venture_Data/projects/riscv/freedom/fpga/e300artydevkit mcs
make[1]: Entering directory '/media/esdentem/Venture_Data/projects/riscv/freedom/fpga/e300artydevkit'
VSRC_TOP=/media/esdentem/Venture_Data/projects/riscv/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v EXTRA_VSRCS="/media/esdentem/Venture_Data/projects/riscv/freedom/rocket-chip/vsrc/AsyncResetReg.v /media/esdentem/Venture_Data/projects/riscv/freedom/rocket-chip/vsrc/DebugTransportModuleJtag.v /media/esdentem/Venture_Data/projects/riscv/freedom/sifive-blocks/vsrc/SRLatch.v" vivado -nojournal -mode batch -source script/board.tcl -source script/prologue.tcl -source script/init.tcl -source script/impl.tcl
awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub

****** Vivado v2016.4 (64-bit)
  **** SW Build 1756540 on Mon Jan 23 19:11:19 MST 2017
  **** IP Build 1755317 on Mon Jan 23 20:30:07 MST 2017
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source script/board.tcl
# set name {arty_e300devkit}
# set part_fpga {xc7a35ticsg324-1L}
# set part_board {digilentinc.com:arty:part0:1.1}
# set bootrom_inst {rom}
source script/prologue.tcl
# set scriptdir [file dirname [info script]]
# set commondir [file dirname $scriptdir]
# set srcdir [file join $commondir src]
# set constrsdir [file join $commondir constrs]
# set wrkdir [file join [pwd] obj]
# set ipdir [file join $wrkdir ip]
# set top {system}
# create_project -part $part_fpga -in_memory
# set_property -dict [list \
#   BOARD_PART $part_board \
#   TARGET_LANGUAGE {Verilog} \
#   SIMULATOR_LANGUAGE {Mixed} \
#   TARGET_SIMULATOR {XSim} \
#   DEFAULT_LIB {xil_defaultlib} \
#   IP_REPO_PATHS $ipdir \
#   ] [current_project]
ERROR: [Board 49-71] The board_part definition was not found for digilentinc.com:arty:part0:1.1. The project's board_part property was not set, but the project's part property was set to xc7a35ticsg324-1L. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Common 17-17] undo 'set_property'

    while executing
"rdi::add_properties -dict {BOARD_PART digilentinc.com:arty:part0:1.1 TARGET_LANGUAGE Verilog SIMULATOR_LANGUAGE Mixed TARGET_SIMULATOR XSim DEFAULT_LI..."
    invoked from within
"set_property -dict [list \
  BOARD_PART $part_board \
  TARGET_LANGUAGE {Verilog} \
  SIMULATOR_LANGUAGE {Mixed} \
  TARGET_SIMULATOR {XSim} \
  DEFAU..."
    (file "script/prologue.tcl" line 12)
INFO: [Common 17-206] Exiting Vivado at Sun Mar 12 15:54:17 2017...
Makefile:13: recipe for target 'obj/system.bit' failed
make[1]: *** [obj/system.bit] Error 1
make[1]: Leaving directory '/media/esdentem/Venture_Data/projects/riscv/freedom/fpga/e300artydevkit'
common.mk:55: recipe for target '/media/esdentem/Venture_Data/projects/riscv/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.mcs' failed
make: *** [/media/esdentem/Venture_Data/projects/riscv/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.mcs] Error 2
make -f Makefile.e300artydevkit mcs  6.51s user 0.12s system 49% cpu 13.299 total
```

Let me know if you need more information.

a0u, 2017-03-13:
Have you installed the [Vivado board files](https://reference.digilentinc.com/reference/software/vivado/board-files) for the Arty?

esden, 2017-03-13:
Thank you this fixes the build issue. It might be a good idea to mention this in the project README, along side with "git submodule update --init --recursive" that is necessary to get all the needed dependencies. :)

Thank you for your quick help!

closed 2017-03-13 by esden

ronak12jain, 2019-06-21:
Hello All,

Guys, Can you please help me with the Vivado version which you are using it?

I am using Vivado 2019 on ubuntu 16.04 and getting few errors when trying to build e300artydevkit bootloader.

The error logs are:
ERROR: [Synth 8-439] module 'BootROM' not found [/home/sysadmin/Ronak/RISC_V_Research/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v:234961]
ERROR: [Synth 8-6156] failed synthesizing module 'TLMaskROM' [/home/sysadmin/Ronak/RISC_V_Research/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v:234888]
ERROR: [Synth 8-6156] failed synthesizing module 'E300ArtyDevKitSystem' [/home/sysadmin/Ronak/RISC_V_Research/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v:327159]
ERROR: [Synth 8-6156] failed synthesizing module 'E300ArtyDevKitPlatform' [/home/sysadmin/Ronak/RISC_V_Research/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v:331529]
ERROR: [Synth 8-6156] failed synthesizing module 'E300ArtyDevKitFPGAChip' [/home/sysadmin/Ronak/RISC_V_Research/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.v:332437]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:33 ; elapsed = 00:01:04 . Memory (MB): peak = 2338.637 ; gain = 623.418 ; free physical = 2384 ; free virtual = 8959
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
230 Infos, 70 Warnings, 0 Critical Warnings and 6 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details

It seems that the problem is related with Vivado version. Also, correct me if I am wrong.

Thanks,
Ronak

erikdanie, 2019-06-24:
I am unable to replicate this issue when using vivado 2018.2. If you do `make -f Makefile.e300artydevkit verilog` to just make the verilog, that step succeeds? but then if you do the mcs step it fails?

ronak12jain, 2019-06-25:
> I am unable to replicate this issue when using vivado 2018.2. If you do `make -f Makefile.e300artydevkit verilog` to just make the verilog, that step succeeds? but then if you do the mcs step it fails?

Hi erikdanie,

Yes, when I run **make -f Makefile.e300artydevkit verilog** then I do not get any error.
But when I try **make-f Makefile.e300artydevkit mcs** then I do get error.

And also the error is related **vivado synthesizing**.

I am using vivado 2019.1 version.

Thanks,
Ronak

ronak12jain, 2019-06-26:
> I am unable to replicate this issue when using vivado 2018.2. If you do `make -f Makefile.e300artydevkit verilog` to just make the verilog, that step succeeds? but then if you do the mcs step it fails?

@erikdanie ,

I have also tried with Vivado 2018.2 and try to build but the result is still same and the error is also same when I build through Vivado 2019.1

When I build for **verilog** it works fine but not in the case of **mcs**.

Can you please provide the steps?
Or is there any dependencies that I need to be taken care of?

Also, I have attached an error logs for your reference.

[vivado_error_log.txt](https://github.com/sifive/freedom/files/3328232/vivado_error_log.txt)

Thanks,
Ronak

erikdanie, 2019-06-26:
Have you edited the code at all? Can you try a clean checkout of freedom and see if that fixes anything?

ronak12jain, 2019-06-27:
> Have you edited the code at all? Can you try a clean checkout of freedom and see if that fixes anything?

@erikdanie ,

To answer your question first, I have not edited the code at all. 
Also, I have tried the fresh checkout freedom repository and build the **mcs** project and now I am able to build successfully. 
To be honest I really don't know how it worked this time. The same command I used to follow which I mentioned in the previous post and it worked this time.

Anyways, now I am able to build mcs project for **e300artydevkit** using **Vivado 2018.2**.

Thanks @erikdanie for your help.

Thanks,
Ronak

shipingli, 2020-11-10:
Any solution after a whole year?