sifive/freedom #21: Unable to access jarfile

trinamic, 2017-03-29:
Hi all,

We just cloned the freedom git repository (worked without errors) and want to compile the verilog sources as described in the readme file using "make -f Makefile.e300artydevkit verilog".
Make is starting but finishes with the error that the sbt-launch.jar ist not accessible.
See the log below.

Kubisch@TMC03 /cygdrive/d/projekte/risc-v/freedom
$ make -f Makefile.e300artydevkit verilog
make -C /cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl SBT="java -jar /cygdrive/d/projekte/risc-v/freedom/rocket-chip/sbt-launch.jar" root_dir=/cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl build-scala
make[1]: Verzeichnis „/cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl“ wird betreten
java -jar /cygdrive/d/projekte/risc-v/freedom/rocket-chip/sbt-launch.jar "assembly"
Error: Unable to access jarfile /cygdrive/d/projekte/risc-v/freedom/rocket-chip/sbt-launch.jar
make[1]: *** [Makefile:23: /cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 1
make[1]: Verzeichnis „/cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl“ wird verlassen
make: *** [common.mk:26: /cygdrive/d/projekte/risc-v/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 2

The file sbt-launch.jar does exist. We clone the repository with all sub-modules recusively. So all required data should be there.
We are using a cygwin terminal on a Windows 10 machine.
Did someone see this error? Any recommendation? Addtional information required?

Thanks
Stephan

PS: not an expert in using make

dbarbi1, 2017-03-31:
Sorry, I am not an expert in Cygwin, but does the Cygwin environment have access to a Java Run Time Environment? For example, from Cygwin if you type java -version, what is the output?

trinamic, 2017-04-03:
Hi,

yes it has access to java.

$ java -version
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.121-b13, mixed mode)

If I go directly into the directory of the  sbt-launch.jar and use the command lines as it is given in the makefile, Java actually seems to get access to sbt-launch.jar and processes something...

java -jar sbt-launch.jar 
[info] Loading project definition from D:\Projekte\RISC-V\freedom\rocket-chip\project
[info] Updating {file:/D:/Projekte/RISC-V/freedom/rocket-chip/project/}rocket-chip-build...
[info] Resolving com.typesafe.sbt#sbt-ghpages;0.5.3 ...
[info] Resolving com.typesafe.sbt#sbt-git;0.6.4 ...
[info] Resolving org.eclipse.jgit#org.eclipse.jgit.pgm;3.3.2.201404171909-r ...
[info] Resolving args4j#args4j;2.0.12 ...
[info] Resolving org.apache.commons#commons-compress;1.6 ...
....

terpstra, 2017-11-03:
I would say the solution to this is to use linux. Running this under windows it not something any of the developers do.

closed 2017-11-03 by terpstra