sifive/freedom #22: Latest E300 Build for Arty Creates a Non-Working Image

donbmouse, 2017-04-27:
The current E300 build succeeds, but doesn't work when flashed to the Arty. Flashing using vivado succeeds without error. Yet, when the Prog button is pushed, the image doesn't work as intended. 

Flashing the 1-0-2 MCS file cited in the Getting Started documentation results in a working E-Series image. So, it doesn't seem to be an issue with the Arty, Vivado, or the flashing process. 

Thanks,
D

mwachs5, 2017-04-27:
If you build the image using this repo, it won't have any executable loaded into the SPI Flash to run. If you want that to happen, you'll need to compile the ELF file from the freedom-e-sdk and load it over the debug interface, or include it as the FLASHED_PROGRAM:
https://github.com/sifive/freedom/blob/f4375c22662f82b1b4f94e88b1aba6998b1f34ba/fpga/e300artydevkit/Makefile#L7

donbmouse, 2017-04-27:
Oh!!! Hah! Thanks, Megan. I knew I must have been missing something. Thanks for your fast response.

closed 2017-04-27 by donbmouse