sifive/freedom #23: Address Map issues

MarcoGeuna, 2017-04-28:
Hi all,

I'm studying the **address map** of the u500vc707devkit. My question is about the Object **ConfigStringOutput**, which you define inside  [freedom/src/main/scala/unleashed/u500vc707devkit/Top.scala#L45](https://github.com/sifive/freedom/blob/master/src/main/scala/unleashed/u500vc707devkit/Top.scala#L45), but here is also a default one created in  [sifive/rocket-chip/src/main/scala/coreplex/RISCVPlatform.scala#L33](https://github.com/ucb-bar/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/coreplex/RISCVPlatform.scala#L33).

**I wonder** if it is only a sort of "graphical feedback" (since, when you compile the Chip, you have output on video) or it is **involved to set the the address map, TileLink or Rocket processor...**

Can you please also clarify the aim of strings **_globalConfigString_** and **_hartConfigStrings_**, which you can find in [TLPLIC](https://github.com/ucb-bar/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/uncore/devices/Plic.scala#L85) or [CoreplexLocalInterrupter](https://github.com/ucb-bar/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/uncore/devices/Prci.scala#L91)?

Thank you so much!

richardxia, 2017-04-28:
FYI, the master branch of rocket-chip has completely replaced ConfigString with Device Tree, which is a standard format for operating systems such as Linux to query for the hardware available on a system. In rocket-chip, the Config String/Device Tree are exposed to hardware via the Boot ROM. I don't have a time frame, but the next major release of `freedom` will use Device Tree.

The `ConfigStringOutput` was a hardcoded string representing the Config String. rocket-chip originally came with a naive implementation of a Config String generator, which would automatically generate a Config String based on the actual configuration of the hardware, but since the Config String specification was never finalized, we used `ConfigStringOutput` as a way to override the generator. This allowed us to at least hardcode a Config String that could be used to boot Linux.

Since Config String has been replaced with Device Tree, rocket-chip will now automatically generate a Device Tree String (DTS) and compile it down to its binary representation (DTB). The `globalConfigString` and `hartConfigStrings` were a way to specifying parts of the Config String. The new Device Tree generator has a much better mechanism for configuring the Device Tree output on a per-hart/per-device basis.

Lastly, the Config String/Device Tree is an *output* of rocket-chip, not an input, so it merely describes the configuration and does not affect the generated hardware.

austinharris, 2017-04-28:
FWIW, I updated freedom [here](https://github.com/austinharris/freedom/tree/priv-1.10) for priv-1.10 and the newer rocket-chip with DTS. I don't actually have the e300 or vc707 to test, but it works with my port to a different FPGA. You can at least see the DTS output if you just want to study the address map.

MarcoGeuna, 2017-05-10:
Thank you for your fast answers! Thank you @richardxia for your explanation, and thank you @austinharris  for making your code available. I really appreciated everything!

In the last few weeks I had to change my priorities, due to a shorter deadline, but now coming back to this with this new information I think (_hope_ ;-) ) I got all the relevant information for me. Thx again!

closed 2017-05-19 by richardxia