sifive/freedom #24: Build error

RISCVevaluation, 2017-05-12:
Hi when I am trying to execute make -f Makefile.e300artydevkit mcs targets in Freedom E 310, i am getting the following error how to i am resolve this issue..... 

# export_ip_user_files -of_objects $obj -no_script -force
invalid command name "export_ip_user_files"
    while executing
"export_ip_user_files -of_objects $obj -no_script -force"
    (file "script/init.tcl" line 36)
INFO: [Common 17-206] Exiting Vivado at Thu May 12 10:42:37 2016...
make[1]: *** [obj/system.bit] Error 1

richardxia, 2017-05-17:
Hi, what version of Vivado do you have installed?

riscveval, 2017-05-19:
Hi, I am using Vivado 2015.1.

richardxia, 2017-05-19:
Vivado 2015.1 doesn't appear to support the `export_ip_user_files` command. Please upgrade to a newer version of Vivado (the WebPACK edition is free and supports the Arty board).

closed 2017-09-22 by mwachs5