sifive/freedom #26: How to add another UART?

lsanmu, 2017-06-09:
Hi all, 
I want to how add another UART device to  U500VC707DevKitSystem.
I know there is a PeripheryUART in U500 code, which the addr is 0x54000000.
If I want to  add another UART device, do I need to implement the another  PeripheryUART  at device directory,  which connect to  peripheryBus? Or, I only need with one more PeripheryUART in U500VC707DevKitSystem?
Everything can be useful! Thank you so much!

mwachs5, 2017-06-09:
You can just add another one to this list:
https://github.com/sifive/freedom/blob/master/src/main/scala/unleashed/u500vc707devkit/Top.scala#L25

mwachs5, 2017-06-09:
As an example, the E300 Dev Kit already has 2 UARTS:
https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/Top.scala#L33

lsanmu, 2017-06-15:
@mwachs5 , Thanks for you useful reply.
And, would you mind give me some advice to simulate the generated verilog code using Cadence ncsim.

closed 2017-08-22 by mwachs5