sifive/freedom #27: Execute in place

MarcoGeuna, 2017-06-17:
Hi all,

I'm trying to understand **what happens during the boot inside u500vc707 system**.

I know that the Rocket core starts execution at address 0x1000, where there is the BootROM. I also know that the BootROM is generated from file [freedom/bootrom/u500vc707devkit.img](https://github.com/sifive/freedom/blob/master/bootrom/u500vc707devkit.img). In the same folder bootrom, I found file [xip/xip.S](https://github.com/sifive/freedom/blob/master/bootrom/xip/xip.S).

Executing xip.S,  the program counter at a certain point performs a [jump to address _XIP_TARGET_ADDR_ (#L19)](https://github.com/sifive/freedom/blob/master/bootrom/xip/xip.S#L19), but **I don't find this constant** defined in the entire repo.

Hence, can you please give me some more information? I fear I am missing something...

Thank you all!

terpstra, 2017-06-19:
The ROM at 0x1000 contains a small program that loads the contents of an SD card to DDR at address 0x8000_0000 and then jumps to it.

MarcoGeuna, 2017-06-20:
OK, than you for this explanation @terpstra!
Let me ask you also this: while doing what you told me, does this small program do some sort of reset operation? Like reset some CPU registers or stuff like that?
I ask you because I am working on a custom version of bootrom.img that mirrors [xip.S](https://github.com/sifive/freedom/blob/master/bootrom/xip/xip.S), and when I jump to memory (the program is already there) the system stops working. I was thinking that maybe, in some ways, the CPU is not ready for doing that, cause of some missing reset... Does it make sense or not at all?

terpstra, 2017-06-20:
I don't think so. At most it might set the UART baud rate. If you are building linux to put in DDR, be aware that the mcs we released uses an older SBI that is incompatible with mainline. You need the version in the freedom-u-sdk/master branch.

austinharris, 2017-06-20:
I'm not too sure about the older rocket-chip that used configstring, but with DTB this bootrom works for me to immediately jump bbl.bin in the DDR:
https://github.com/austinharris/freedom/blob/vu190/bootrom/u500vu190devkit.S

In short: it doesn't appear you need to do anything to reset the CPU and the uart should start working once Linux starts to output.

Do you have your code publicly available?

MarcoGeuna, 2017-07-02:
Hi all! Thank you for you answers!

I finally ended up finding the solution to my problems: the simulation (I am using Mentor Graphics Modelsim) was not working cause I didn't add the verilog define RANDOMIZE_*, like [here](https://github.com/freechipsproject/rocket-chip/blob/master/vsim/Makefrag#L53). Since, during simulation, there were so many conflicts between signals (in 'X' state), I though about a wrong reset signal and, from that, a wrong bootROM code. But now this error seems to be solved.

Coming to you questions, @austinharris , I am using the older rocket-chip with configstring and I'm running bare metal tests, without OS support. My source code is not online, but I post here below the one I use to create the bootROM. As I already told, it mirrors xip/xip.S.

Thx for everything guys!

-----------------------------------------------------------------------------------------------------------------------------------
My xip.S source code:

```
// See LICENSE for license details.
// Execute in place
// Jump directly to DRAM_BASE
#define DRAM_BASE 0x80000000

  .text
  .option norvc
  .globl _start
_start:
  j 1f
  nop
  nop
#ifdef CONFIG_STRING
  .word cfg_string
#else
  .word 0  // Filled in by GenerateBootROM in Chisel
#endif

1:
  li t0, DRAM_BASE
  jr t0
  nop
  nop
  nop
  nop
  nop

  .section .rodata
#ifdef CONFIG_STRING
cfg_string:
  .incbin CONFIG_STRING
#endif
```

closed 2017-09-22 by mwachs5