sifive/freedom #28: “$ make -f Makefile.e300artydevkit verilog”  error

Pengnian, 2017-06-28:
hi,all 

when i use this command to generted verilog code, met jvm error! please help me~thanks.

[image]

richardxia, 2017-06-28:
Have you tried increasing the size of the perm gen space using extra command line flags to the JVM?

You may want to replace [this line](https://github.com/sifive/freedom/blob/f4375c22662f82b1b4f94e88b1aba6998b1f34ba/common.mk#L19) of the Makefile with:

```makefile
SBT ?= java -XX:MaxPermSize=256M -jar $(rocketchip_dir)/sbt-launch.jar
```

Also, apparently Java 8 [removed permanent generation](https://stackoverflow.com/questions/18339707/permgen-elimination-in-jdk-8), so you will never run out of PermGen space on Java 8.

Pengnian, 2017-06-29:
@richardxia  Thanks Richard,

Both add your code in common.mk and replace java7 with java 8 all work, but another question show  again!
[image]

richardxia, 2017-06-29:
It seems that you must be using a very old version of Scala that isn't properly parsing `build.sbt`. Apparently, [three years ago](https://stackoverflow.com/questions/22890829/play-framework-error-parsing-expression-in-build-sbt), you needed to put blank lines between each line. Can you try upgrading Scala to `2.11.7` to match `rocket-chip` (but definitely not to `2.12`)?

Pengnian, 2017-06-29:
Hi, 

my scala version is 2.11.7,so met that question! where can download old version of scala?

[image]

richardxia, 2017-06-29:
I think that shows the version of Scala sbt will use to compile the project, but I don't think that is the version of Scala that is used to read in the sbt project files. Perhaps you could try installing Scala manually from the official website? https://www.scala-lang.org/download/2.11.7.html

closed 2017-09-22 by mwachs5