sifive/freedom #29: The board_part definition was not found for digilentinc.com:arty:part0:1.1

Pengnian, 2017-06-28:
hi,all

cd  freedom/fpga/e300artydevkit
make 

[image]

mwachs5, 2017-06-28:
You need to install the board part for the Arty board. There are instructions here on Digilent's website: https://reference.digilentinc.com/reference/software/vivado/board-files

Pengnian, 2017-06-28:
@mwachs5  Thank you very much, I met a trouble again ! As a  beginner, Thanks your patience!
[image]

Pengnian, 2017-06-28:
@mwachs5 
[image]

richardxia, 2017-06-28:
It looks like the `VSRC_TOP` environment variable isn't getting set for some reason. Can you post the first few lines of console output immediately after you run `make -f Makefile.e300artydevkit mcs`, but before Vivado starts running?

Pengnian, 2017-06-29:
@richardxia  Hi Richard,

when I  run  `make -f Makefile.e300artydevkit mcs`, doesn't work also!
[image]

mwachs5, 2017-07-03:
This second error looks the same as your other issue #28. Please check the version of scala you are using, it seems you are using a very old version.

Pengnian, 2017-07-10:
@mwachs5 Okay,Thanks your remind !

closed 2017-09-22 by mwachs5