sifive/freedom #30: use "freedom-e-sdk" to upload and debug my programs,but met some errors.

Pengnian, 2017-07-09:
Hi,all 
I'm use command `make upload PROGRAM=hello BOARD=freedom-e300-arty` in the `freedon-e-sdk `directory. so,you can tell  me   the why?
[image]

mwachs5, 2017-07-10:
Please check your wiring connections to the debugger. It looks like you may have swapped or missing some connections.

Pengnian, 2017-07-11:
@mwachs5 Thank you, the LED of debugger is light when connection between board and computer .
But when I use command `make upload PROGRAM=hello BOARD=freedom-e300-arty`,the led not bright ! But every time the errors maybe difference! 
[image]

mwachs5, 2017-07-11:
What FPGA image are you using? The message is telling you the JTAG ID is 0x20000001, which sounds like the Coreplex IP eval kit, not the Freedom E300 Arty Dev Kit. You should use BOARD=coreplexip-... to match

Pengnian, 2017-07-19:
@mwachs5  Thank you very much !

Pengnian, 2017-07-19:
@mwachs5 
[image]

It's okay?

closed 2017-08-22 by mwachs5

zhvn100, 2020-08-14:
Hello，i met the same problem . i use arty a7 fpga. The connection between board and the debugger  and  the situation of led of debugger are the same as you. But my error always as follows:
[image]
So,did you figure it out finally ?  please tell me about it , thank you