sifive/freedom #31: "run synthesis" in the vivado2016.4  "module AsynResetReg not found"

Pengnian, 2017-07-12:
Hi,all

At first, I want to know if  the generated verilog code by  command `$ make -f Makefile.e300artydevkit verilog`  just is RISC-V core not contain demo program ?

Then,  my fpga  image  contain  fpga configuration file and demo program ,  so  when I upload the  program into board by ` freedom-e-sdk`  occur  some errors , whether there is relationship with that ?

Finally,I want to generate verilog according the command of `$ make -f Makefile.e300artydevkit verilog` . But, after generated verilog code add to vivado 2016.4 , when "run synthesis" occured some error.The prompt message is "module AsyncResetReg not found",I don't konw how to solve it , please tell me if you know. Thanks very much!

[image]

Pengnian, 2017-07-12:
Debug errors by `make upload PROGRAM=hello BOARD=freedom-e300-arty` in directory of `freedom-e-sdk`. By the way, the  fpga image  contain fpga configuration file and  demo program.

[image]

mwachs5, 2017-07-12:
These are two different issues. In the second issue it looks like your FPGA is still programmed with Coreplex IP, not the Freedom E300 Dev Kit, so you need to use BOARD=coreplexip-e31-arty (or e51-arty if that's what you have on there).

Pengnian, 2017-07-13:
@mwachs5 Thank you,

closed 2017-08-22 by mwachs5