sifive/freedom #32: Arty E300 flow broken

z419379295, 2017-08-20:
Hi,

I noticed that the repo has been updated recently and a pull request is accepted, and the flow of generating mcs for Arty board is broken.

What I did:

git clone --recursive https://github.com/sifive/freedom.git
cd freedom
make -f Makefile.e300artydevkit verilog (as per README) - this generates verilog without error.
make -f Makefile.e300artydevkit mcs (as per README) - this prompts an error below:

"

make -C /home/ken/freedom/bootrom/xip romgen
make[1]: Entering directory '/home/ken/freedom/bootrom/xip'
dtc -I dts -O dtb -o /home/ken/freedom/builds/e300artydevkit/E300ArtyDevKitConfig.dtb /home/ken/freedom/builds/e300artydevkit/E300ArtyDevKitConfig.dts
/bin/riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -O2 -std=gnu11 -Wall -I. -nostartfiles -fno-common -g -DXIP_TARGET_ADDR=0x20400000 -DDEVICE_TREE='"/home/ken/freedom/builds/e300artydevkit/E300ArtyDevKitConfig.dtb"' -static -nostdlib -o /home/ken/freedom/builds/e300artydevkit/xip.elf xip.S
make[1]: /bin/riscv64-unknown-elf-gcc: Command not found
Makefile:17: recipe for target '/home/ken/freedom/builds/e300artydevkit/xip.elf' failed
make[1]: *** [/home/ken/freedom/builds/e300artydevkit/xip.elf] Error 127
make[1]: Leaving directory '/home/ken/freedom/bootrom/xip'
common.mk:68: recipe for target '/home/ken/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.rom.v' failed
make: *** [/home/ken/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.rom.v] Error 2

"

Is this intentional? How could I resolve this? Thanks.

ss2783, 2017-08-20:
You need to set the environment variable for $RISCV which points to the location where your local version of the RISC-V toolchain is installed to compile the bootloaders.

z419379295, 2017-08-20:
Thanks. It works now.

z419379295, 2017-08-22:
Sorry for opening this issue again, I got another error when I try to generate the mcs file as per README:

"
File "/projects/freedom0821/freedom/rocket-chip/scripts/vlsi_rom_gen", line 114
for key, val in iterate_by_n(line.split(), 2)}
^
SyntaxError: invalid syntax
make1: *** /projects/digital/work/ken/freedom0821/freedom/builds/e300artydevkit/rom.v Error 1
make1: Leaving directory `/projects/freedom0821/freedom/bootrom/xip'
make: *** /projects/freedom0821/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.rom.v Error 2
"

Did I miss anything (package, library etc)? Thanks.

mwachs5, 2017-08-22:
What python version are you using? I just tested this on Python 2.7.12.

```
>> python --version
Python 2.7.12
```

z419379295, 2017-08-23:
Aha, that solves the problem. Thanks.

closed 2017-08-23 by z419379295