sifive/freedom #34: [Synth 8-439] module 'mmcm' not found

Pengnian, 2017-09-25:
Hi SiFive's expert,

$ make -f Makefile.e300artydevkit verilog 

When I adding generated verilog code to Vivado 2016.4  and synthesis it encounter error ,please tell me how to fix it. Thanks~
[image]

richardxia, 2017-09-25:
You need to generate the Xilinx [Mixed-Mode Clock Manager (MMCM) IP](https://www.xilinx.com/products/intellectual-property/mmcm_module.html) yourself if you're not using our Makefile scripts to synthesize the FPGA image. Here is a guide on how to do that in the Vivado GUI: http://users.wpi.edu/~rjduck/MMCM%20Vivado%20example%20Verilog.pdf

You can find the parameters that we used in https://github.com/sifive/fpga-shells/blob/2389e6e95717caca782e7444422da16fef687188/xilinx/arty/tcl/ip.tcl#L3-L13

Pengnian, 2017-09-27:
Hi Richard,
       
        The  Xilinx ip "mmcm"and "reset_sys" has added project but where is ip of "PowerOnResetFPGAOnly " I dont know ! Please help me ~ 

Thank  you  very !

Pengnian, 2017-09-27:
@richardxia  Thanks  Richard,
The xilinx ip "mmcm"and "reset_sys" can find  IP Catalog but "PowerOnResetFPGAOnly " I  don't know can find where or need to write myself ? Please give me  some hints~  
 Thank you again!
[image]

mwachs5, 2017-09-27:
https://github.com/sifive/fpga-shells/tree/master/xilinx/common/vsrc

Pengnian, 2017-09-27:
@mwachs5  Hi mwach,
I know some module under directory of `freechipsproject/rocket-chip/vsrc` and `sifive/sifive-block` but where is `BootRom  module` , I cant  find it !
[image]

richardxia, 2017-09-27:
BootROM uses Chisel's sequential memory feature, so it does not refer to an existing Verilog module. You will need to use the `vlsi_rom_gen` script from rocket-chip to generate the Verilog file from the ROM parameters. See our Makefile rule for handling this: https://github.com/sifive/freedom/blob/22ee433699dca058f96858c15e7a2b0c33547d63/bootrom/xip/Makefile#L38

terpstra, 2017-11-03:
Looks like this is resolved.

closed 2017-11-03 by terpstra