sifive/freedom #35: PPA Comparison

riscveval, 2017-09-28:
Hi I am evaluating various RISCV Cores. I am getting the following results for Sifive E31 & E51. 

Core | ISA | Bit | Pipeline | LUTs | LUT RAMs | LUT FFs | BRAM | Power µW/MHz | Dhrystone DMIPS/   MHz | Coremark /MHz | Target FPGA
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
SiFive E51 | RV64IMAC | 64 | 6 | 52655 | 4492 | 45160 | 18 | 0.038 | 1.80 | 2.76 | VC707
SiFive E31 | RV32IMAC | 32 | 5 | 13458 | 328 | 7688 | 5 | 0.012 | 1.61 | 2.73 | Arty

Is these results are correct or not? Can you share these information to me.
I want to change the cache memory size for Sifive Cores. So can you share how to change the cache memory for E31 & E51.

terpstra, 2017-11-03:
The Dhrystone DMIPS/MHz seem right. I can't comment on the rest.

IC-Dream, 2018-06-01:
@riscveval Hi, when you make verilog for E51, have you met the following error:
[error] /root/freedom/sifive-blocks/src/main/scala/devices/mockaon/MockAONPeriphery.scala:7:8: object HasPeripheryCLINT is not a member of package freechips.rocketchip.devices.tilelink
[error] import freechips.rocketchip.devices.tilelink.HasPeripheryCLINT
[error] ^
[error] /root/freedom/sifive-blocks/src/main/scala/devices/mockaon/MockAONPeriphery.scala:16:35: not found: type HasPeripheryCLINT
[error] trait HasPeripheryMockAON extends HasPeripheryCLINT with HasPeripheryDebug { this: BaseSubsystem =>
[error] ^

Could you find type HasPeripheryCLINT in the path:rocket-chip/src/main/scala/devices/tilelink/CLINT.scala??? i only can find the "CanHavePeripheryCLINT" in the CLINT.scala in my local. but it contains the HasPeripheryCLINT in github repo. it's strange.

Thanks for your help

chlizheng, 2018-08-30:
@riscveval 
Hi,
How do you evaluate  the dhrystone of various RISCV cores? Do youo have the general  bench? Now I also want to evaluate some riscv cores. Could you give me your bench of evaluation of dhrystone?

brucehoult, 2018-08-30:
@chlizheng freedom-e-sdk contains RISC-V gcc compiler and other things for building programs for SiFive embedded RISC-V cores. It comes with configuration files (e.g. "make" include file setting the ISA and ABI, linker script) for various SiFive-supported SoC and FPGA boards. It should be possible to add configuration files for almost any other RISC-V embedded core.

You can find it at https://github.com/sifive/freedom-e-sdk

chlizheng, 2018-08-31:
@brucehoult 
Thanks,I'll try it.

chlizheng, 2018-09-13:
Hi, now, I want to evaaluate the performance of myself riscv processor. And I use "riscv32-unknown-elf-gcc -march=rv32em -mabi=ilp32e benchmark.c" and "riscv32-unknown-elf-objdump -dr --disassembler-options=numeric,no-aliases a.out >benchmark.S" to get the instruction file and then plan to  run it  in myself processor. But  when I check  a.out and benchmark.S I find that the start address of a.out is 0x0000 but it's 0x10074 in benchmark.S. Besides, the hex value of instrustions in the two files can't match. For example, I can't find the value of a.out in the benchmark.S. Is my method wrong or the command having problems?
best wishes!

jim-wilson, 2018-09-13:
Objdump is a pretty simple tool.  It is unlikely that the output is wrong.

Why do you think that the start address for the a.out is 0x0?  "objdump -f a.out" will give you the program start address.    "objdump -p a.out" will show you the ELF segment info.  You should see something like
    LOAD off    0x00000000 vaddr 0x00010000 paddr 0x00010000 align 2**12
         filesz 0x00001e54 memsz 0x00001e54 flags r-x
which says that file offset 0 gets loaded at memory offset 0x10000.  The first part of the ELF file is elf headers, and if the elf headers are 0x74, then the first instruction is at 0x10074.  Programs are typically not loaded at address 0, because that breaks handling for null pointers.

You have the instruction encoding, 00e78c63.  You shouldn't need the raw offset.  But if you really do, then you can manually decode the instruction, or you can compute it by subtracting the instruction address from the target address.  The objdump output is meant to be human readable, and printing the target address not the offset is most useful to humans.

IC-Dream, 2018-09-14:
@jim-wilson please help to check this issue, it seems that the output of objdump is wrong.
https://github.com/riscv/riscv-tests/issues/162

chlizheng, 2018-09-14:
@jim-wilson 
Thanks, I have misunderstanded before.

chlizheng, 2018-09-14:
@jim-wilson 
I also have some questions now. I have try to compile many .c files and all of them memory offset are 0x10000.Is the 0x10000 stable? Can the memory offset 0x10000 be changed and configurable? And for the instruction "100cc: 00e78c63 beq x15,x14,100e4 <deregister_tm_clones+0x28>" , the right formal of it in the spec is similar" 100cc: 00e78c63 beq x15,x14,0x18"? Is there a way to get  this kind of instrustion by toolchain rather than compute manually? Besides, I think a.out  contains many other information data besides instrution data such as PC value.Can I just get the logic value of instruction? Maybe my comprehend for a.out is wrong.
best wishes!

jim-wilson, 2018-09-14:
The 0x10000 comes from the default linker script.  You can see the default linker script by running the linker with --verbose, e.g. "riscv32-unknown-elf-ld --verbose".