sifive/freedom #37: E300 with Hardware accelerator

NaumanWazir, 2017-10-03:
Hi, I am trying to already present hardware accelerator in Rocket chip. To use it I am doing the following step:
1. In freedom/rocket-chip/src/main/scala/coreplex/Config.scala adding this 
class WithAccumRocc extends Config((site, here, up) => {
  case RocketTilesKey => up(RocketTilesKey, site) map { r =>
    r.copy(rocc =
      Seq(
        RoCCParams(
          opcodes = OpcodeSet.custom1,
          generator = (p: Parameters) => LazyModule(new CharacterCountExample()(p)))
        ))
    }
})
2. In freedom/rocket-chip/src/main/scala/system/Config.scala adding this 
class TinyConfig extends Config(
//add by me   
  new WithAccumRocc ++
  new WithNMemoryChannels(0) ++
  new WithStatelessBridge ++   
  new WithNTinyCores(1) ++
  new BaseConfig)
 Now when I built it in freedom using make -f Makefile.e300artydevkit verilog I am getting the following error:
java -jar /home/anbuntu/freedom/rocket-chip/sbt-launch.jar "run-main freechips.rocketchip.system.Generator /home/anbuntu/freedom/builds/e300artydevkit sifive.freedom.everywhere.e300artydevkit E300ArtyDevKitFPGAChip sifive.freedom.everywhere.e300artydevkit E300ArtyDevKitConfig"
[info] Loading project definition from /home/anbuntu/freedom/rocket-chip/project
Using addons: 
[info] Set current project to freedom (in build file:/home/anbuntu/freedom/)
[info] Compiling 1 Scala source to /home/anbuntu/freedom/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Packaging /home/anbuntu/freedom/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Compiling 1 Scala source to /home/anbuntu/freedom/rocket-chip/target/scala-2.11/classes...
[info] Running freechips.rocketchip.system.Generator /home/anbuntu/freedom/builds/e300artydevkit sifive.freedom.everywhere.e300artydevkit E300ArtyDevKitFPGAChip sifive.freedom.everywhere.e300artydevkit E300ArtyDevKitConfig
[info] [0.003] Elaborating design...
Interrupt map (1 harts 52 interrupts):
  [1, 2] => aon
  [3, 3] => uarts_0
  [4, 4] => uarts_1
  [5, 5] => qspis_0
  [6, 6] => spis_0
  [7, 7] => spis_1
  [8, 39] => gpios_0
  [40, 43] => pwms_0
  [44, 47] => pwms_1
  [48, 51] => pwms_2
  [52, 52] => i2c_0

[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:50)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:238)
	at chisel3.Driver$.elaborate(Driver.scala:92)
	at freechips.rocketchip.util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:53)
	at freechips.rocketchip.system.Generator$.elaborate(Generator.scala:12)
	at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:100)
	at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
	at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
	at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:106)
	at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
	at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:89)
	at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[... 108 lines cut ...]
	at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
	at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:89)
	at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at freechips.rocketchip.system.Generator$.main(Generator.scala:12)
	at freechips.rocketchip.system.Generator.main(Generator.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[trace] Stack trace suppressed: run last freedomPlatforms/compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last freedomPlatforms/compile:runMain for the full output.
[error] (freedomPlatforms/compile:runMain) Nonzero exit code: 1
[error] Total time: 11 s, completed 3 Oct, 2017 9:22:50 AM
common.mk:48: recipe for target '/home/anbuntu/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.fir' failed
make: *** [/home/anbuntu/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.fir] Error 1
But when I built TinyConfig in freedom/rocket-chip/vsim using make verilog CONFIG=TinyConfig there is no error and I could see the code is generated with accelerator. I am banging my head on this for a month now but can figure out what I am doing wrong. Can you please help me?

colinschmidt, 2017-10-04:
I have reproduced this and am looking at why the interaction between RoCC and the e300 causes this error.

terpstra, 2017-11-03:
@colinschmidt any updates?

NaumanWazir, 2017-11-05:
Hi, I have narrow down the problem to Platform.scala. I am new to this so don't know the actual reason. But if you comment out E300ArtyDevKitPlatformIO and define pins inside E300ArtyDevKitPlatform like this 
  val io = new Bundle {
    val jtag_reset = Bool(INPUT)
    val ndreset    = Bool(OUTPUT)
    val jtag = new JTAGPins(() => PinGen(), false)
    val gpio = new GPIOPins(() => PinGen(), p(PeripheryGPIOKey)(0))
    val qspi = new SPIPins(() => PinGen(), p(PeripherySPIFlashKey)(0))
    val aon = new MockAONWrapperPins()
  }
Then you can built it. I checked it on Arty it run demo_gpio but gives error 03 when I try to access accelerator. I am looking into it.

colinschmidt, 2017-11-07:
When I was looking at this it seemed like a chisel problem and I was seeing if a bump would fix the problem (obviously that was non-trivial). I'll check up on that sometime this week

aignacio, 2018-11-05:
Hello @colinschmidt,
Did you get the reason for **error 3** like @NaumanWazir mentioned? I could build with WithRoccExample but then when I try to access it happens the same...
[image]