sifive/freedom #38: Correct `freedom-u-sdk` branch / known-good SD Card image for U500 VC707?

ilebedev, 2017-10-16:
I am having a whole lot of trouble finding a working combination of hardware and SD card image for the u500 on a VC707. By way of a sanity check, I am trying to get vanilla hardware and software to boot.

I have in the past successfully done this with older versions of the repository.

I have tried the following hardware:
- [latest, Aug 20](sifive/freedom-u-sdk) hardware built with vivado 2016.4
- pre-built MCS [linked](https://static.dev.sifive.com/freedom-u500-vc707-mcs-0-1.zip) in the u500 vcu707 guide.

And the following software
- some older vanilla `freedom-u-sdk` I had on an SD from a few months ago
- Unmodified build of the [latest, Dec 2](https://github.com/sifive/freedom-u-sdk/commit/b38f7c98485e78b9638a04e499d01fcb8ebf2958) `freedom-u-sdk`, `master` branch.
- Unmodified build of the [latest, Jun 27](https://github.com/sifive/freedom-u-sdk/commit/deabfb7b14c441740872a8674aab37d3316b4447) `freedom-u-sdk`, `priv-1.10` branch.

In all cases, the result is the same: the bootloader runs, but fails to boot linux:

```
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT

```

followed by nothing. I suspect some sort of disagreement between the priv ABI enforced by hardware, and expected by software.

These builds take a long time; any pointers would be greatly appreciated.
Is there a pre-built, known-good SD card image somewhere? This would sure help reduce the search space for this issue. Please advise.

Thanks in advance,
-ilia

shhssdm, 2017-10-18:
I'm having exactly the same problem. I was trying to backtrace the problem and found something is going wrong at riscv-pk/machine/minit.c line #27 with the old vanilla branch. The statement makes u500 to print something but at the end it hangs at do_tohost_fromhost where you cannot escape the infinite loop. Don't know the reason.

saurabhpb22, 2017-10-20:
I too am having the same issue although I am trying to boot with out the FMC PCI board. I have made changes accordingly to work without the FMC but I get stuck at the same state as the above prints. I kept doubting my changes (as there was no way for me to test the base code) however thanks to this post I think there might be some issue with base SiFive code.

terpstra, 2017-11-02:
@palmer-dabbelt @henrystyles Is the freedom-u-sdk/master in-sync with the (very old) vc707 image on our site? We should probably also provide a newer vc707 image compatible with the latest (hopefully final) revision of the privileged spec...

palmer-dabbelt, 2017-11-02:
Ya, we need to bump everything.  If freedom/master is updated then I can go deal with the software.

terpstra, 2017-11-02:
I think the main effort required is to create a new VC707 image using only open-source stuff.

ilebedev, 2017-11-02:
I am unable to get work done for nearly a month now because of this issue. Happy to build the images myself, but unclear which `freedom-u-sdk` is compatible with recent commits to `freedom`. 

Is there a specific `freedom-u-sdk` commit that is known to boot on 22ee433699dca058f96858c15e7a2b0c33547d63 in `freedom`? If not, perhaps a specific earlier earlier commit to `freedom`? Trying pairs until a working one is found is _extremely_ time consuming and a huge waste of time if somoene on this list knows the answer.. Surely someone here knows the answer.

Any info would help here, thanks,

closed 2017-11-03 by terpstra

terpstra, 2017-11-04:
freedom/master should work with freedom-u-sdk/master now. Furthermore, we are preparing an updated freedom-u-sdk which should better support FPGAs.

terpstra, 2017-11-07:
> Hi @terpstra <https://github.com/terpstra>, thanks for the update.
>
> A small correction, in case anyone is stuck: freedom/master successfully
> boots with freedom-u-sdk/priv-1.10. (/new immediately panics, and /master
> has seen no updates and exhibits the same opaque failure to boot).

We will bump the sdk and remove those branches this week. Hopefully, from
that point on they will stay compatible since the hardware-softeware
interface is no longer changing.

PR #40 (2017-11-03): Bump all hardware to the newest versions
This PR, combined with a new freedom-[ue]-sdk, should bring the hardware and software used for FPGA builds in-sync.

Fixes #38.