sifive/freedom #55: ”make -f Makefile.u500vc707devkit verilog“ error

IC-Dream, 2018-05-29:
Hi,

when i want to make verilog for FPGA platform, got the fllowing error:

make -C /mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl SBT="java -jar /mnt/hgfs/SharedDir/freedom/rocket-chip/sbt-launch.jar" root_dir=/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl build-scala
make[1]: Entering directory `/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl'
java -jar /mnt/hgfs/SharedDir/freedom/rocket-chip/sbt-launch.jar "assembly"
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/project
[info] Loading settings from build.sbt ...
[info] Set current project to firrtl (in build file:/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/)
[info] Updating ...
[info] Done updating.
[error] java.io.FileNotFoundException: /mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/target/streams/$global/assemblyOption/$global/streams/assembly/c25a67d8a2a44bb463d039a0572114b81c8c0d54_dir/firrtl/passes/CommonSubexpressionElimination$$anonfun$firrtl$passes$CommonSubexpressionElimination$$eliminateNodeRef$1$1.class (File name too long)
[error] 	at java.io.FileOutputStream.open0(Native Method)
[error] 	at java.io.FileOutputStream.open(FileOutputStream.java:270)
[error] 	at java.io.FileOutputStream.<init>(FileOutputStream.java:213)
[error] 	at java.io.FileOutputStream.<init>(FileOutputStream.java:162)
[error] 	at sbt.io.Using$.$anonfun$fileOutputChannel$1(Using.scala:93)
[error] 	at sbt.io.Using$$anon$2.openImpl(Using.scala:76)
[error] 	at sbt.io.OpenFile.open(Using.scala:43)
[error] 	at sbt.io.OpenFile.open$(Using.scala:39)
[error] 	at sbt.io.Using$$anon$2.open(Using.scala:75)
[error] 	at sbt.io.Using$$anon$2.open(Using.scala:75)
[error] 	at sbt.io.Using.apply(Using.scala:21)
[error] 	at sbt.io.IO$.$anonfun$copyFile$3(IO.scala:714)
[error] 	at sbt.io.IO$.$anonfun$copyFile$3$adapted(IO.scala:713)
[error] 	at sbt.io.Using.apply(Using.scala:22)
[error] 	at sbt.io.IO$.copyFile(IO.scala:713)
[error] 	at sbt.io.IO$.copyImpl(IO.scala:651)
[error] 	at sbt.io.IO$.$anonfun$copy$1(IO.scala:639)
[error] 	at scala.Function$.$anonfun$tupled$1(Function.scala:77)
[error] 	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[error] 	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:52)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[error] 	at scala.collection.TraversableLike.map$(TraversableLike.scala:227)
[error] 	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error] 	at sbt.io.IO$.copy(IO.scala:639)
[error] 	at sbt.io.IO$.copyDirectory(IO.scala:685)
[error] 	at sbt.io.IO$.copyDirectory(IO.scala:674)
[error] 	at sbt.io.IO$.copyDirectory(IO.scala:657)
[error] 	at sbtassembly.Assembly$.$anonfun$assembleMappings$8(Assembly.scala:194)
[error] 	at scala.collection.parallel.AugmentedIterableIterator.map2combiner(RemainsIterator.scala:112)
[error] 	at scala.collection.parallel.AugmentedIterableIterator.map2combiner$(RemainsIterator.scala:109)
[error] 	at scala.collection.parallel.immutable.ParVector$ParVectorIterator.map2combiner(ParVector.scala:62)
[error] 	at scala.collection.parallel.ParIterableLike$Map.leaf(ParIterableLike.scala:1052)
[error] 	at scala.collection.parallel.Task.$anonfun$tryLeaf$1(Tasks.scala:49)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.control.Breaks$$anon$1.catchBreak(Breaks.scala:63)
[error] 	at scala.collection.parallel.Task.tryLeaf(Tasks.scala:52)
[error] 	at scala.collection.parallel.Task.tryLeaf$(Tasks.scala:46)
[error] 	at scala.collection.parallel.ParIterableLike$Map.tryLeaf(ParIterableLike.scala:1049)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute(Tasks.scala:149)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute$(Tasks.scala:145)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.compute(Tasks.scala:440)
[error] 	at java.util.concurrent.RecursiveAction.exec(RecursiveAction.java:189)
[error] 	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:289)
[error] 	at java.util.concurrent.ForkJoinTask.doJoin(ForkJoinTask.java:389)
[error] 	at java.util.concurrent.ForkJoinTask.join(ForkJoinTask.java:719)
[error] 	at scala.collection.parallel.ForkJoinTasks$WrappedTask.sync(Tasks.scala:375)
[error] 	at scala.collection.parallel.ForkJoinTasks$WrappedTask.sync$(Tasks.scala:375)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.sync(Tasks.scala:440)
[error] 	at scala.collection.parallel.ForkJoinTasks.executeAndWaitResult(Tasks.scala:423)
[error] 	at scala.collection.parallel.ForkJoinTasks.executeAndWaitResult$(Tasks.scala:414)
[error] 	at scala.collection.parallel.ForkJoinTaskSupport.executeAndWaitResult(TaskSupport.scala:56)
[error] 	at scala.collection.parallel.ExecutionContextTasks.executeAndWaitResult(Tasks.scala:555)
[error] 	at scala.collection.parallel.ExecutionContextTasks.executeAndWaitResult$(Tasks.scala:555)
[error] 	at scala.collection.parallel.ExecutionContextTaskSupport.executeAndWaitResult(TaskSupport.scala:80)
[error] 	at scala.collection.parallel.ParIterableLike$ResultMapping.leaf(ParIterableLike.scala:956)
[error] 	at scala.collection.parallel.Task.$anonfun$tryLeaf$1(Tasks.scala:49)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.control.Breaks$$anon$1.catchBreak(Breaks.scala:63)
[error] 	at scala.collection.parallel.Task.tryLeaf(Tasks.scala:52)
[error] 	at scala.collection.parallel.Task.tryLeaf$(Tasks.scala:46)
[error] 	at scala.collection.parallel.ParIterableLike$ResultMapping.tryLeaf(ParIterableLike.scala:951)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute(Tasks.scala:149)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute$(Tasks.scala:145)
[error] 	at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.compute(Tasks.scala:440)
[error] 	at java.util.concurrent.RecursiveAction.exec(RecursiveAction.java:189)
[error] 	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:289)
[error] 	at java.util.concurrent.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1056)
[error] 	at java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1692)
[error] 	at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:157)
[error] (assembly / assembledMappings) java.io.FileNotFoundException: /mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/target/streams/$global/assemblyOption/$global/streams/assembly/c25a67d8a2a44bb463d039a0572114b81c8c0d54_dir/firrtl/passes/CommonSubexpressionElimination$$anonfun$firrtl$passes$CommonSubexpressionElimination$$eliminateNodeRef$1$1.class (File name too long)
[error] Total time: 30 s, completed 2018-5-29 15:03:48
make[1]: *** [/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 1
make[1]: Leaving directory `/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl'
make: *** [/mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 2

Thanks a lot for your help

brucehoult, 2018-05-29:
java.io.FileNotFoundException: /mnt/hgfs/SharedDir/freedom/rocket-chip/firrtl/target/streams/$global/assemblyOption/$global/streams/assembly/c25a67d8a2a44bb463d039a0572114b81c8c0d54_dir/firrtl/passes/CommonSubexpressionElimination$$anonfun$firrtl$passes$CommonSubexpressionElimination$$eliminateNodeRef$1$1.class (File name too long)

I make that file path (rather than file name as such, which is 112) 281 characters long.
Those are both fine on Linux (ext2/ext3/ext4/zfs anyway).

I don't *think* Java itself has a file name or path length limit.

I believe NTFS has a limit of 260 characters for the path.

Your "/mnt/hgfs" makes me suspect you're running a Linux virtual machine under Windows, with an NTFS partition mounted there.

It would probably be better if you had mentioned this at the start (assuming it is true).

IC-Dream, 2018-05-30:
@brucehoult Thanks a lot, i did the "make verilog" under /mnt/hgfs/SharedDir which is a shared dir between windows and linux.

closed 2018-05-30 by IC-Dream