sifive/freedom #62: /bin/riscv-64-unknown-elf-gcc: command not found

jjtoms4s, 2018-06-06:
this error happened when i generate the mcs file
i have compile the toolchain once in the rock-chip just follow the commands in this link https://github.com/freechipsproject/rocket-chip#setting-up-the-riscv-environment-variable
do i have to compile again in the /freedom/rocket-chip/riscv-tools or i can use the former generated toolchain,if so where can i find it.
by the way,i used to compile the riscv-toolchain use the command: ./build-rv32ima.sh does this means i generate a file named riscv-32-unknown-elf-gcc.but what i need is riscv-64-unknown-elf-gcc.

jjtoms4s, 2018-06-06:
i set up the riscv enviroment with this command: export RISCV=/path/to/install/riscv/toolchain
and it worked. it seems i used the former built toolchain,but does it ok to use this toolchain.
can anyone explain the difference between the different versions of riscv-toolchain/gcc i m a little confused

closed 2018-06-06 by jjtoms4s

nanoeng, 2020-07-24:
Had a similar issue and script was reporting the following messages

> ./riscv-tools/build.sh: line 6: build.common: No such file or directory
> Starting RISC-V Toolchain build process
> ./riscv-tools/build.sh: line 19: build_project: command not found
> ./riscv-tools/build.sh: line 21: build_project: command not found
> ./riscv-tools/build.sh: line 22: build_project: command not found
> ./riscv-tools/build.sh: line 23: build_project: command not found
> ./riscv-tools/build.sh: line 24: build_project: command not found
> ./riscv-tools/build.sh: line 25: build_project: command not found

After some digging, tried the following steps and got it going

1. Set up RISCV as described above
1. Run build.sh from path/to/freedom/rocket-chip/riscv-tools (for any reason, the script does not show error/warning messages when run at a different location). This step should take a little while (be suspicious if it doesn't) 
1. issues & errors might arise and would require[ GCC is installed piece by piece](https://stackoverflow.com/questions/9450394/how-to-install-gcc-piece-by-piece-with-gmp-mpfr-mpc-elf-without-shared-libra)
1. If errors show up, install libgmp-dev and libmpc (using 'sudo apt-get install libmpc-dev')