sifive/freedom #66: Arty E300 MCS Generation Error

smiqbal, 2018-06-07:
I'm getting an error when I try to generate Verilog files for the E300.
`make -f Makefile.e300artydevkit verilog` runs successfully, but running `make -f Makefile.e300artydevkit mcs` gives me the following errors:

`make -C /home/anon/freedom/bootrom/xip romgen
make[1]: Entering directory '/home/anon/freedom/bootrom/xip'
dtc -I dts -O dtb -o /home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.dtb /home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.dts
make[1]: dtc: Command not found
Makefile:10: recipe for target '/home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.dtb' failed
make[1]: *** [/home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.dtb] Error 127
make[1]: Leaving directory '/home/anon/freedom/bootrom/xip'
common.mk:68: recipe for target '/home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.rom.v' failed
make: *** [/home/anon/freedom/builds/e300artydevkit/sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitConfig.rom.v] Error 2`

I suspect this is because all the submodules were not cloned successfully. Running `git submodule update --init --recursive --progress` gives me the following errors:

`Cloning into '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/dtc'...
fatal: unable to connect to git.qemu-project.org:
git.qemu-project.org[0: 172.99.69.163]: errno=Connection timed out

fatal: clone of 'git://git.qemu-project.org/dtc.git' into submodule path '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/dtc' failed
Failed to clone 'dtc'. Retry scheduled
Cloning into '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/pixman'...
fatal: unable to connect to anongit.freedesktop.org:
anongit.freedesktop.org[0: 131.252.210.161]: errno=Connection timed out
anongit.freedesktop.org[1: 2610:10:20:722:a800:ff:fe24:61cf]: errno=Network is unreachable

fatal: clone of 'git://anongit.freedesktop.org/pixman' into submodule path '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/pixman' failed
Failed to clone 'pixman'. Retry scheduled
Cloning into '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/roms/SLOF'...
fatal: unable to connect to git.qemu-project.org:
git.qemu-project.org[0: 172.99.69.163]: errno=Connection timed out

fatal: clone of 'git://git.qemu-project.org/SLOF.git' into submodule path '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/roms/SLOF' failed
Failed to clone 'roms/SLOF'. Retry scheduled
Cloning into '/home/anon/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/roms/ipxe'...`

I've already tried editing the git config file `/freedom/.git/modules/rocket-chip/modules/riscv-tools/modules/riscv-gnu-toolchain/config` to replace the git protocol with https as follows

`url = https://github.com/riscv/riscv-qemu.git` from `git://github.com/riscv/riscv-qemu.git` but that did not change anything. These submodules appear to be unreachable from looking at the repository itself. Any help would be greatly appreciated.

jim-wilson, 2018-06-07:
It is a problem on the qemu site.  Adding something like this to your ~/.gitconfig file should work around it.

[url "https://git.qemu.org/git"]
insteadOf = https://git.qemu-project.org

Actually you are using the git protocol, so that probably needs to be git instead of https in the above two lines.  Or maybe try adding two copies, one for git and one for https.

See for instance
    https://github.com/riscv/riscv-gnu-toolchain/issues/280
    https://github.com/riscv/riscv-gnu-toolchain/issues/324
For more info and suggestions.

smiqbal, 2018-06-07:
@jim-wilson Is that the same file as `~/freedom/.git/modules/rocket-chip/modules/riscv-tools/modules/riscv-gnu-toolchain/config`?

Currently `~/freedom/.git/modules/rocket-chip/modules/riscv-tools/modules/riscv-gnu-toolchain/config` has this line:
`[submodule "riscv-qemu"]`
`       url = https://git.qemu.org/git`

That does not appear to change where the git tries to fetch the files from as I'm still getting errors of the form:

`fatal: clone of 'git://git.qemu-project.org/dtc.git' into submodule path '/home/syediqbal/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/dtc' failed`
`Failed to clone 'dtc'. Retry scheduled`
`Cloning into '/home/syediqbal/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/pixman'...
fatal: unable to connect to anongit.freedesktop.org:`
`anongit.freedesktop.org[0: 131.252.210.161]: errno=Connection timed out`
`anongit.freedesktop.org[1: 2610:10:20:722:a800:ff:fe24:61cf]: errno=Network is unreachable`

`fatal: clone of 'git://anongit.freedesktop.org/pixman' into submodule path '/home/syediqbal/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/pixman' failed`
`Failed to clone 'pixman'. Retry scheduled`
`Cloning into '/home/syediqbal/freedom/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-qemu/roms/SLOF'...`
`fatal: unable to connect to git.qemu-project.org:`
`git.qemu-project.org[0: 172.99.69.163]: errno=Connection timed out`

Thanks for the help. I don't have experience changing gitconfig files so I'm not sure where I should add the lines you have mentioned.

jim-wilson, 2018-06-07:
The file that you edit is the .gitconfig file in your home directory.  The two lines I mentioned worked for me.  One of the issues I linked to mentioned a slightly different solution that worked for them.

You could alternatively try changing the ..gitmodules file inside the riscv-qemu dir, but there are multiple levels of submodules, both in the github tree and the qemu-project.org tree, and I'm not sure if changes to the github .gitmodules file will actually work.

It might be simpler if you try checking out riscv/riscv-qemu, and once you get that working, then you can checkout bigger packages that include riscv-qemu.

smiqbal, 2018-06-07:
Thanks for your help @jim-wilson. Adding this to my ~/.gitconfig fixed the issue:

`[url "https://git.qemu.org/git"]`
`insteadOf = https://git.qemu-project.org`

`[url "https://git.qemu.org/git"]`
`insteadOf = git://git.qemu-project.org`

`[url "https://github.com/"]`
`insteadOf = git://github.com/`

`[url "https://github.com/"]`
`insteadOf = git@github.com:`

`[url "https://anongit.freedesktop.org/git"]`
`insteadOf = git://anongit.freedesktop.org`

closed 2018-06-07 by smiqbal