sifive/freedom #70: demo_gpio jtag upload  failed

jjtoms4s, 2018-06-11:
[image]
i run into this error when i upload software,does anyone knows ( i used the pre-generated mcs file to generate the hardware in fpga so the circuit is no problem )

aignacio, 2018-09-04:
Hello @jjtoms4s, seeing your image seems that you are trying to open another instance of openocd but It's busy (because the session is still open...). Also, are you using the JTAG from OIimex as they recommend? if you are using another JTAG adapter tool, remember to change the device descriptor and the VID/PID to your correspondent tool...

jjtoms4s, 2018-09-04:
hellow i receive the email and thanks for your reply, i ve solved this issue , just forget to close this issue

closed 2018-09-04 by jjtoms4s