sifive/freedom #72: Porting E300 to platform without quad SPI flash

teadotjay, 2018-06-28:
I'm trying to port the E300 to the VCU108, which does not include a quad SPI flash memory like the Arty. Is there a configuration guide that can help get me started on bypassing the flash boot and executing bare metal code out of a block memory?

brucehoult, 2018-06-28:
Have you read https://www.sifive.com/documentation/freedom-soc/freedom-e300-arty-fpga-dev-kit-getting-started-guide/ ?

teadotjay, 2018-06-28:
Yes, and the section on customizing the FPGA image is a single paragraph that points to this site. I've also read UCB/EECS-2016-17 and tried to follow the links (most are broken). I found another FPGA implementation at https://github.com/ucb-bar/fpga-zynq, but this indicates it is deprecated and does not use the latest Rocket core. I can't find where the configuration parameters are documented, and I can't find any examples that use block memory for program code.

mhanuel26, 2018-08-01:
Hi @teadotjay ,

I think it would be difficult to find documentation, actually I have done a few [attempts ](https://github.com/sifive/freedom/issues/46) and it seems no one knows about it or there  is no interest at all. 
Going towards a totally different platform will be even harder as you are totally alone that route.

Cheers,

mhanuel26, 2018-08-02:
@mwildoer Very interesting and rich amount of knowledge, thank you for sharing.
I will read carefully.  When you say the project is in beta stage, at this moment how much can I try with available downloads?

mwildoer, 2018-08-02:
In the blog, you'll see a section about the RISC-V processor from Kristian Skordal (https://github.com/skordal/potato). It explains how to get this working on the Arty board, but this can be modified for your VCU108 FPGA board, as it's all open (the main differences, I guess, would be the constraints file for pinouts).

There are also instructions in the blog about how to set up the gcc compiler, so you can compile the sample C code and get the uart, etc. working.

Hopefully this should enable you to get the Potato RISC-V processor working on your board, as most of the work is already done.

If you subscribe to my blog, you'll see when I release more info, including the Intel Hex loader, IDE, sprite engine, etc. I'd be happy to update the IDE to work for your FPGA board as well, as it should work through the uart in a similar way to the Arty.

mhanuel26, 2018-08-02:
@mwildoer actually is @teadotjay the owner of a  VCU108.

My interest is because I have an Arty-S7 board which is somehow different from Arty-35T, and my last attempt was not possible because the Arty-S7 FPGA have a different way to generate the QSPI clock, I think your point about the VHDL source code visibility of E300 is one of the things that make it more difficult.

One last question please, do you think it's possible to get [Ada Risc-V](https://blog.adacore.com/ada-on-the-first-risc-v-microcontroller) compiler running on this core?

mwildoer, 2018-08-02:
Ada compilation would be great! At the moment, even for C, the UART drivers in the Potato RISC-V SoC sample code are very basic, and this functionality would have to be ported to Ada (to get the terminal working, etc).

Since the HiFive1 board is already supported by AdaCore, it does sound doable. Also, I saw that the SiFive gcc tools now support Ada, so I could add Ada support to RVStudio (syntax highlighting, compilation, etc).

Do you think you could get very simple UART code working in Ada, similar to the current C code example? If you could, I think Ada support could happen...

mhanuel26, 2018-08-02:
Hi @mwildoer ,

I think I could do it. 
I will study things a little bit more and come back here. 
My first thing to do is to make the RVStudio work for my Arty S7 board, which I think is not big deal since no special difference I have seen after reviewing the arty.xdc file in potato example folder.

Adding Ada support to RVStudio is difficult? 
I guess you are concern about UART since it's somehow the way to test the port is working,  right?