sifive/freedom #78: Why doesn't write after read hazard in E51/U54

IC-Dream, 2018-08-09:
Hi,

I noticed something different from other in-order CPU cores in SiFive FU549-C000 Manual v1p0 on page 17, mentioned that the pipeline only interlocks on RAW and WAW, why doesn't write after read hazard in E51/U54?
[image]

Thanks
Dream

brucehoult, 2018-08-09:
The E31 manual (e.g. HiFive1) says the same thing.
Maybe this is common to all Rocket-based designs, but I don't know. I'm guessing there is some kind of store buffer.

IC-Dream, 2018-08-10:
As far as I know, for In-order cpu, WAR/WAW/RAW hazards are natural, we can't aviod them, since it is different from out-of-order CPU, which could use rename to eliminate the WAR/WAW hazards.

brucehoult, 2018-08-10:
@aswaterman do the various loads and stores (including CSR) all have single-cycle throughput, despite the multi-cycle latency to use the result?
I'm assuming MUL/DIV/RAM have lower throughput as well as longer latency.
It might be worth being explicit about throughput vs latency in the manual.

aswaterman, 2018-08-10:
The manual already has some of this in other places ("The multiplier can only execute one operation at a time and will block until the previous operation completes").  It's not generally possible to assign a numeric value, since these events' latencies are data-dependent and can depend on factors outside the core complex.

brucehoult, 2018-08-10:
Loads and stores to addresses present in cache/TIM? And CSRs .. can you issue multiple reads to different CSRs and wait for all the results, or does each one stall?

aswaterman, 2018-08-10:
Yes, loads and stores are fully pipelined, assuming cache hits.  The manual implies this, but doesn't clearly state it.  Same for CSR reads.

IC-Dream, 2018-08-10:
@aswaterman Since the E51/U54 are single issue, the continuous write are always at different stage, So the WAW hazard should not manifest?

IC-Dream, 2018-08-10:
got it, Thanks so much.

closed 2018-08-10 by IC-Dream