sifive/freedom #88: vera iofpga build fails

tmagik, 2018-09-04:
Both of the following fail on master:
```
make -f Makefile.veraiofpga verilog
make -f Makefile.veraiofpga prjx
```
The first should be an easy fix, the second requires some documentation and regular regression tests on a system with Microsemi Libero tools installed. (and/or some docs on how to get the tools set up on linux)

Going back to commit d548b5fa61729de5b6467931d80501409f4ed358 results in a working 'make verilog' target, however I currently have no way to test if that build produces a working bitstream.

terpstra, 2018-09-04:
The whole flow for libero is broken at the moment.

tmagik, 2018-09-06:
The following sequence worked, so I apparently can't reproduce the previous failure condition I observed:
```
 1203  git clone https://github.com/sifive/freedom freedom-2
 1204  cd freedom-2/
 1205  ls
 1206  git submodule update --init --recursive && make -f Makefile.veraiofpga verilog
```

It seems like this is due to some state from running a previous make or partial checkout (without 'submodule update --init --recursive' which has now been cleared

tmagik, 2018-09-06:
So the actual reproducible problem (once you have a working licensed libero installation) is:

```
*****************************************************************
******************** Building Libero project ********************
*****************************************************************
Number of arguments: 5
Chisel build directory: /home/tmagik/freedom-2/builds/vera-iofpga
Chisel model: IOFPGAChip
Chisel project: sifive.freedom.unleashed.vera.iofpga
Chisel config: IOFPGAConfig
Chisel board: vera
FlashPro Express folder: /home/tmagik/freedom-2/builds/vera-iofpga/FlashProExpress
Info: The PolarFire family only supports the enhanced constraint flow.
The IOFPGAChip project was created.
---------- Executing Libero TCL script: /home/tmagik/freedom-2/builds/vera-iofpga/sifive.freedom.unleashed.vera.iofpga.IOFPGAConfig.Libero.polarfire_tx_pll.tcl ----------
Error:  Cannot find Spirit core configuration file for vendor:Actel library:SgCore name:PF_TX_PLL version:1.0.109.
Error:  The command 'create_design' failed.
Error:  Failure when executing Tcl script. [ Line 111 ]
Error:  The Execute Script command failed.
The IOFPGAChip project was closed.
```

tmagik, 2018-09-07:
So the first problem is the if you do a new installation of Libero, it may have new(er) versions of cores like PF_TX_PLL, which is now version 1.0.112.

If I work through those errors, I end up with the following modified TCL script:

```
puts "||||||| pcie_root_port "
new_file -type {SmartDesign} -name polarfire_pcie_rp
puts "||||||| add_vlnv_instance "
add_vlnv_instance -component {polarfire_pcie_rp} -library {} -vendor {Actel} -lib {SgCore} -name {PF_PCIE} -version {1.0.242} -instance_name {PF_PCIE_0} -promote_all 0 -file_name {}

puts "|||||||| pcie_root_port start configure_vlnv"
configure_vlnv_instance -component {polarfire_pcie_rp} -library {} -name {PF_PCIE_0} -params {"UI_IS_CONFIGURED:true"} -validate_rules 0
```
which produces this error:
```
---------- Executing Libero TCL script: /home/tmagik/freedom-2/builds/vera-iofpga/sifive.freedom.unleashed.vera.iofpga.IOFPGAConfig.Libero.polarfire_pcie_root_port.tcl ----------
||||||| pcie_root_port
||||||| add_vlnv_instance
Info:  Group pin `PF_PCIE_0:PADs_IN` cannot be promoted because it has pins which are already grouped on the top level.
Info:  PAD pin `PF_PCIE_0:PADs_IN` of `PF_PCIE_0` is automatically connected to a top-level port.
Info:  Group pin `PF_PCIE_0:PADs_OUT` cannot be promoted because it has pins which are already grouped on the top level.
Info:  PAD pin `PF_PCIE_0:PADs_OUT` of `PF_PCIE_0` is automatically connected to a top-level port.
Info:  The connection is being dropped from the net 'PCIE_0_WAKE_N' because the pin 'PCIE_0_WAKE_N' does not exist.
Error:  Design rules check of "polarfire_pcie_rp" failed.
        Check the message window for more information.
        
         [OK]
Error:  The command 'generate_design' failed.
Error:  Failure when executing Tcl script. [ Line 111 ]
Error:  The Execute Script command failed.
```

tmagik, 2018-09-14:
It turns out that the current fpga-shells project only supports [Libero 2.1](https://www.microsemi.com/document-portal/doc_download/1243245-download-libero-soc-polarfire-v2-1-for-linux).

Other notes are that the script to start 'synplify_pro' needs to be changed from #!/bin/sh to #/bin/bash (on debian 9)

CLappin, 2019-01-24:
Hi @terpstra and @tmagik,

Sorry for piggy backing on this thread but I'm noticing a similar issue when building the vera io-fpga.

**repo:**
    https://github.com/sifive/freedom

**Command used:** 
    make -f Makefile.veraiofpga verilog

**Error seen:**
    Caused by: java.lang.IllegalArgumentException: requirement failed: There is no TLError with    Acquire reachable from ChipLink. One must be instantiated.

This is from a clean clone and initialization of the freedom github page. 
I tried the e300artydevkit and it built without any issues. 

**Additional testing:**
I've also tried another commit (b1ebb204fa8216b4a1ef225adab172777be7af51ues) which I was able to build the vera verilog code from successfully.

Thanks,
Ciaran

ronak12jain, 2019-06-19:
Hello All,

I wanted to build vera iofpga but getting an error.

The commands I have followed:
git clone https://github.com/sifive/freedom
cd freedom
git submodule update --init --recursive
make -f Makefile.veraiofpga verilog

When I tried to build e300artydevkit then it works successfully.
make -f Makefile.e300artydevkit verilog

Any suggestions for vera iofpga???

Thanks,
Ronak

erikdanie, 2019-06-19:
Can you post the error? We are in the process of overhauling the iofpga builds, but it should work fine off master

ronak12jain, 2019-06-20:
Hi @erikdanie ,

Thanks for your response.

I have attached a file contains a build logs for vera iofpga.

Thanks,
Ronak
[report.log](https://github.com/sifive/freedom/files/3308755/report.log)

erikdanie, 2019-06-20:
Oh ok, I see. This error is being fixed. You can try checking out the branch freedom-fix, doing a git submodule update and making that. I tested this locally and make verilog works. I did not test make prjx though, and note that you will need at least Libero 12.0 to make prjx on that branch.

This branch is in the final stages of release candidate testing and will be merged to master soon

ronak12jain, 2019-06-21:
Hi @erikdanie ,

I have followed the steps you mentioned, first I have updated freedom core with "git submodule update --init --recursive " and then tried to build veraiofpga by following command "sudo make -f Makefile.veraiofpga verilog" but the result is still same getting an error.

Also, I am attaching an error logs for your reference.

Is there any dependencies to build veraiofpga that I need to install/update first?

Thanks,
Ronak
[report1.log](https://github.com/sifive/freedom/files/3313008/report1.log)

mcd500, 2019-06-21:
Have you tried the command line bellow?
```
git clone https://github.com/sifive/freedom.git
cd freedom
git checkout freedom-fix <- this is required to try the freedom-fix branch
git submodule update --init --recursive
```
and then start build.

ronak12jain, 2019-06-25:
> and then start build

Hi mcd500,

I do able to build veraiofpga by the given commands.

Thanks,
Ronak