sifive/freedom #89: Info about rocket core build

aignacio, 2018-09-10:
Hello all,
I'm getting started in the design of the hardware of E300 (Arty dev. kit) but I didn't understand exactly which file configure and setup the Rocket processor subsystem because in the file **System.scala** from where we call Rocket there's no information about DTIM, bus width and other parameters that I would like to change to see how it affects the size of the FPGA image.
[image]
Also, as far as I understood the configurable options are those available at freedom/rocket-chip/src/main/scala/subsystem/Configs.scala but which of these were select for the default rocket core? And there is some file that prints a resume of what's had been build inside the SoC after scala compilation....Thanks in advance and sorry for the long newbie question  =)

KevinLTSiFive, 2019-02-18:
I think what you want is the top-level config file. It is here in the following link
https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/Config.scala
"TinyConfig" defines the usage of DTIM and you can change it to other configs.

aignacio, 2019-02-22:
Hi @KevinLTSiFive thanks for the reply but now I'm quite more familiarized with the RocketChip project so than I'm able to change the specs as I want, anyway thanks for suggestion.

closed 2019-02-25 by aignacio