sifive/freedom #94: bring up linux on vcu118

shawnofthedead, 2018-11-12:
Hi all,

I would like to bring up linux on vcu118. However, I was unable to do so and  I couldn't find any documentation on such matter. Please help me check my procedure.  

I generated mcs + prm file using Makefile.u500vcu118devkit and Vivado 2018.1. Do I need to update to Vivado 2018.2? 

After I got mcs + prm file, I followed sifive_u500_vc707_gettingstarted guide as a template and hoped its procedure has resemblance to that of vcu118. According to Chapter 3,  I chose "mt28gu01gaax1e-bpi-x16" as my memory part and programed my fpga board. However, it gave me the following console message:

Unspecified failure
ERROR: [Labtools 27-3144] Invalid option: Unspecified failure

Due to this failure, I chose "mt25qu01g-spi-x1_x2_x4" instead after some research. This time around, I was to flash program successfully on to the fpga board.  Is this the correct memory part to use? 
Another issue I have is for SW16 in vcu118 board, is "0001" the correct setting for sifive linux bringup?

As for linux image, I used the bbl file generated by project "freedom-u-sdk". I used the same SD card with bbl in it to successfully bring up linux in my vc707 board. However, when moving to vcu118, I could not get anything to show up in my TeraTerm terminal. 

I understand at the early stage there is no user guide for vcu118. If anyone successfully bring up linux on vcu118 board, please help me debug my issue. 

Thanks

farzadfch, 2018-11-12:
As for the platform flash, you need to choose mt25qu01g-spi-x1_x2_x4_x8.

shawnofthedead, 2018-11-13:
Thanks for your input.

I switch to mt25qu01g-spi-x1_x2_x4_x8 instead. However, that doesn't seem to fix the issue.

farzadfch, 2018-11-16:
Are you connecting FMC and PCIE edge adapters? If not, you need to remove them as it is explained [here](https://forums.sifive.com/t/vcu118-nvdla-implementation/1679/2).

shawnofthedead, 2018-11-20:
May I ask what FMC and PCIE edge adapters you are using, i.e. module names?

closed 2018-11-29 by shawnofthedead

austinharris, 2019-02-27:
Hello,

I am trying to bring this up on my VCU118 board as well, but am having trouble with the sd-card boot over the pmod connector.

I created the sd-card using bbl.bin from the current master of freedom-u-sdk (should this work or is an older version needed?). I directly wrote bbl.bin to the sd-card, rather than using the freedom-u-sdk flow which partitions it (this is only for the real hifive unleashed if I understand correctly). Whenever I reset the cpu on the FPGA I get an error when reading data from the sd card:

`INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  sd_cmd: timeout
ERROR
`

I tried different sd-cards and some give different results (e.g., failing on some of the earlier commands rather than the read, failing due to CRC instead of timeout). Are there certain specs that the sd-card must have in order to work correctly with the bootrom for the vcu118? If someone has a link to a known-working sd-card that would be helpful.

@shawnofthedead could you please describe if you were able to get this working and if there were any additional steps besides removing the FMC and PCIE and connecting the sd-card over pmod?

Thanks!
Austin

shawnofthedead, 2019-02-27:
I did manage to get it work. I also remembered dealing with similar time-out errors.
To solve the time out error, you need vcu118 SCUI and CP210x windows driver installed. 
Then in SCUI, 

1. Go to About -> Restore Board Defaults. (This set board back to its original state, eliminating any unwarranted power/voltage change)

2. Go to FMC -> Set VADJ -> Current/Boot-up (both) -> Set VADJ 1.8 V ( I usually set both just to be safe, however maybe only set boot-up is sufficient)

Hopefully this help.

K16DIABLO, 2020-07-17:
Hi, I have a problem when booting the linux on vcu118. Unlike previous comments, it seems I cannot even enter the bootloader.

[image]

What should I do to enter the bootloader. I connected Pmod sd and configured sw16 to 0001 correctly.