sifive/freedom #95: IntelliJ build

Mixermachine, 2018-12-06:
Hello there :)

I'm writing my bachelor thesis about a RISC-V topic (https://groups.google.com/a/groups.riscv.org/forum/#!topic/isa-dev/vf33dd6ulh0) and are currently digging threw the code.

I like IntelliJ pretty much so is it possible to import this project in IntelliJ and build it?
I get the following errors when IntelliJ executes its automated 
->
syncing...
   dumping project structure from sbt
<-
step:
```console
rror while importing sbt project:

[info] Loading settings from idea.sbt ...
[info] Loading global plugins from /home/e6430/.sbt/1.0/plugins
[info] Loading project definition from /home/e6430/risc-v/freedom/project
[info] Loading settings from build.sbt ...
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/e6430/risc-v/freedom/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons:
[info] Set current project to freedom (in build file:/home/e6430/risc-v/freedom/)
[info] sbt server started at local:///home/e6430/.sbt/1.0/server/1f89f27f0539ea652908/sock
sbt:freedom>
[info] Defining Global / sbtStructureOptions, Global / sbtStructureOutputFile, shellPrompt
[info] The new values will be used by no settings or tasks.
[info] Reapplying settings...
Using addons:
[info] Set current project to freedom (in build file:/home/e6430/risc-v/freedom/)
[info] Applying State transformations org.jetbrains.sbt.CreateTasks from /home/e6430/.IntelliJIdea2018.2/config/plugins/Scala/launcher/sbt-structure-1.0.jar
[info] Reapplying settings...
Using addons:
[info] Set current project to freedom (in build file:/home/e6430/risc-v/freedom/)
[info] Updating ProjectRef(uri("file:/home/e6430/risc-v/freedom/rocket-chip/"), "rocketchip")...
[warn] Choosing local for edu.berkeley.cs#chisel3_2.12;3.2-SNAPSHOT
[warn] 	module not found: edu.berkeley.cs#hardfloat_2.12;1.2
[warn] ==== local: tried
[warn]   /home/e6430/.ivy2/local/edu.berkeley.cs/hardfloat_2.12/1.2/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/hardfloat_2.12/1.2/hardfloat_2.12-1.2.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/e6430/.sbt/preloaded/edu.berkeley.cs/hardfloat_2.12/1.2/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/e6430/.sbt/preloaded/edu/berkeley/cs/hardfloat_2.12/1.2/hardfloat_2.12-1.2.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#hardfloat_2.12;1.2: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
...```

Seems to me that it tries to get hardfloat build with scala version 2.12 but the version in the project only builds with version 2.11.12 (and is setup with this version).
When I run it with the included make file everything compiles fine.

Can someone help me/clear up the situation?
Can I setup IntelliJ to build the sources (and all sub sources) correctly?

Using IntelliJ Ultimate (student version) btw.

Thanks in advance,

Aaron

jackkoenig, 2018-12-06:
You need to override the Scala version, if you look at how SBT is invoked by the [Makefile]( https://github.com/sifive/freedom/blob/3624efff1819e52cec30c72f9085158189f8b53f/common.mk#L31), it includes `++2.12.4`.

When you run the sbt interactive shell, you just need to make `++2.12.4` your first command. Also see this StackOverflow post: https://stackoverflow.com/questions/52049463/rebased-and-now-facing-scala-dependency-issues

Mixermachine, 2018-12-08:
Thank you so much jack :)
I'm baffled this is not mentioned in the setup.
Code highlighting and compilation are working now.

The " dumping project structure from sbt" step still spits some errors:
a required artifact is not listed by module descriptor: *#*!*.*
(multiple times)

Is this fixable?

jackkoenig, 2018-12-09:
If you would like to summarize the steps you needed to get IntelliJ working, we would appreciate contributions to the setup instructions.

Per that error, I don't know of any way to fix it; if you figure it out, please let us know!

Mixermachine, 2018-12-13:
I did create a small PR: https://github.com/sifive/freedom/pull/98
Thanks for your help.

I tried to fiddel with the settings and got the " dumping project structure from sbt" error to disappear.

Thanks again for your help.

closed 2018-12-13 by Mixermachine