sifive/freedom #105: How many resources does the freedom U500 use?

hyf6661669, 2019-02-20:
Unfortunately I can't afford a VC707 board. At present I only have a Genesys2 board with Xilinx FPGA XC7K325T-2FFG900C. I'm wondering whether I can put the freedom U500 into it. Can anyone tell me some information? Much thanks!

brucehoult, 2019-02-20:
200k LUTs and 1 GB RAM and 2 MB block RAM .. not bad.

We've got a single core U500 fitting into an Arty 100T. Logically, your FPGA should be able to fit at *least* a two core U500.

I believe the biggest issue with different Xilinx 7-series boards, given that we already support several of them, is simply getting the CPU core etc wired up to appropriate I/O on that board. I think someone who knows what they are doing with FPGAs (it's not me :( ) should be able to go into the fpga-shells/xilinx directory and clone the subdirectory for the nearest match and edit the config.

farzadfch, 2019-02-20:
Have you considred using [FireSim](https://github.com/firesim/firesim)? If your purpose is to experiment with Rocket Chip and you do not need physical access to the board, that's I think your best option.

hyf6661669, 2019-02-21:
> Have you considered using [FireSim](https://github.com/firesim/firesim)? If your purpose is to experiment with Rocket Chip and you do not need physical access to the board, that's I think your best option.

@farzadfch 
Thanks for telling me about FireSim, I know nothing about that berfoe. I will have a look at it.

> 200k LUTs and 1 GB RAM and 2 MB block RAM .. not bad.
>
> We've got a single core U500 fitting into an Arty 100T. Logically, your FPGA should be able to fit at _least_ a two core U500.
>
> I believe the [...]

@brucehoult 
Thanks, I will try to modify the constraint files for my board.

closed 2019-02-21 by hyf6661669