sifive/freedom #108: freedom: how to build VCU118 FPGA with edge_xdma from fmc_xdma settings?

shawnofthedead, 2019-03-16:
Currently, we have successfully built VCU118 FPGA with PCIe RC mode (fmc_xdma). Now we are switching to PCIe Endpoint mode (edge_xdma), however with no success. What is the best way to build VCU118 with PCIe in Endpoint mode? Thanks 

@erikdanie

erikdanie, 2019-03-16:
@henrystyles

henrystyles, 2019-03-18:
Hi @shawnofthedead
 
The edge_xdma that you see here 
https://github.com/sifive/fpga-shells/blob/9eb56625ba252e6abbb7ded5158331b7da7ea39a/src/main/scala/shell/xilinx/VCU118NewShell.scala#L230
is intended to provide PCIe Root Complex operation using the VCU118 PCIe finger connection + one of these :
https://www.xilinx.com/products/boards-and-kits/1-fd4b4j.html
Currently there is no support for endpoint mode.