sifive/freedom #110: Question about branch prediction

AlexChenzixuan, 2019-03-21:
hello，sorry to bother you . 
  I have read the SIFIVE E20 Core Complex Manual, i found that your branch prediction strategy is  conditional branches are predicted not-taken. I want to konw the reason why you chose this strategy instead of  the backward branches will be taken and the forward branches will not. 
  I am looking forward to your reply！

aswaterman, 2019-03-21:
Predicting backwards branches taken is more accurate, but in order to act on a taken prediction, you need to know the branch target address.  In pipelines where the branch target address can be computed before the branch direction, or in pipelines where employing a branch-target-address predictor is a good use of resources, predicting backwards branches taken will usually improve performance over the predict-not-taken scheme.

In the E20, the branch target and the branch direction are computed in the same clock cycle, so without a branch-target-address predictor, it doesn't make sense to predict branches taken.  And because the E20 is optimized for low area, and because the taken-branch penalty is so low (usually just 1 cycle), the E20 does not include a branch-target-address cache.

In other words, always predicting branches not-taken is a good strategy in this particular case, but that does not generalize to deeper pipelines.

AlexChenzixuan, 2019-03-21:
Okay，got it!  Thank you very much.

closed 2019-03-21 by aswaterman