sifive/freedom #111: build fix for vc707 with latest Vivado

mcd500, 2019-03-23:
It is in the README.md that the `This flow requires Vivado 2016.4. Newer versions are known to fail.`
but I was still trying the build on vc707 with latest vivado and the bellow patch was able to make me to build bitstream successfully.
It was failing that the latest vivado do not have `-version 2.8 -name axi_pcie` and it has 2.9, so I just omitted specifying the version.
Would someone like to comment of the patch bellow.

$ git diff
diff --git a/src/main/scala/ip/xilinx/vc707axi_to_pcie_x1/vc707axi_to_pcie_x1.scala b/src/main/scala/ip/xilinx/vc707axi_to_pcie_x1/vc707axi_to_pcie_x1.scala
index 1884238..5adb9ba 100644
--- a/src/main/scala/ip/xilinx/vc707axi_to_pcie_x1/vc707axi_to_pcie_x1.scala
+++ b/src/main/scala/ip/xilinx/vc707axi_to_pcie_x1/vc707axi_to_pcie_x1.scala
@@ -401,7 +401,7 @@ class VC707AXIToPCIeX1(implicit p:Parameters) extends LazyModule
   ElaborationArtefacts.add(
     "vc707axi_to_pcie_x1.vivado.tcl",
     """ 
-      create_ip -vendor xilinx.com -library ip -version 2.8 -name axi_pcie -module_name vc707axi_to_pcie_x1 -dir $ipdir -force
+      create_ip -vendor xilinx.com -library ip -name axi_pcie -module_name vc707axi_to_pcie_x1 -dir $ipdir -force
       set_property -dict [list \
       CONFIG.AXIBAR2PCIEBAR_0             {0x40000000} \
       CONFIG.AXIBAR2PCIEBAR_1             {0x00000000} \

mcd500, 2019-03-23:
This is the build/
```bash
$ ls -la builds/vc707-u500devkit/obj
total 931776
drwxr-xr-x 4 vc707 vc707      4096  3月 20 15:25 .
drwxr-xr-x 7 vc707 vc707      4096  3月 20 15:25 ..
drwxr-xr-x 5 vc707 vc707      4096  3月 20 13:33 ip
-rw-r--r-- 1 vc707 vc707  52345044  3月 20 14:57 post_opt.dcp
-rw-r--r-- 1 vc707 vc707  77737516  3月 20 15:11 post_place.dcp
-rw-r--r-- 1 vc707 vc707 102319948  3月 20 15:20 post_route.dcp
-rw-r--r-- 1 vc707 vc707  53025937  3月 20 14:55 post_synth.dcp
drwxr-xr-x 2 vc707 vc707      4096  3月 20 15:25 report
-rw-r--r-- 1 vc707 vc707  20273545  3月 20 15:21 VC707Shell.bit
-rw-r--r-- 1 vc707 vc707  55756924  3月 20 15:25 VC707Shell.mcs
-rw-r--r-- 1 vc707 vc707       423  3月 20 15:25 VC707Shell.prm
-rw-r--r-- 1 vc707 vc707 485689638  3月 20 15:23 VC707Shell.sdf
-rw-r--r-- 1 vc707 vc707 106953277  3月 20 15:23 VC707Shell.v
```

tmagik, 2019-03-23:
This looks good to me, Erik do you think this poses any issues?

erikdanie, 2019-03-25:
2017 versions just didn't work at all, but it seems like they may have fixed whatever issue we had in 2018.2. I will test this a bit but assuming all is good, we should do this update

nandakrishnan16, 2019-06-21:
Does this also work with the 2019.1 version?

mcd500, 2019-06-21:
@nandakrishnan16 I have the vc707 at the office so I can not check it until next week, but if you could try it, I would really appreciate your outcome.

Iripi97, 2019-08-15:
This was successful for me using a VM with CentOS 6.10 which had Vivado 2018.3 installed on it.