sifive/freedom #116: Instructions for vc707 without PCIe

mcd500, 2019-04-13:
This is not an issue, it is request for comments.

Do you mind anybody trying the branch in my repo for no pcie on vc707?
I am planning to make a pull request when I have time to bump them up to latest in master.

The source and instruction of building Rocket for vc707 with no-pcie.
The readme:
https://github.com/mcd500/freedom/tree/vc707-nopci
To build,
```
git clone https://github.com/mcd500/freedom.git
cd freedom
git submodule update --init --recursive --force
git checkout -b vc707-nopci remotes/origin/vc707-nopci
export RISCV=/home/riscv/riscv64-elf-tc <- please change it accordingly to your env
export PATH=${PATH}:/opt/Xilinx/Vivado/2016.4/bin <- please change it accordingly to your env
make -f Makefile.vc707-u500devkit nopcie
make -f Makefile.vc707-u500devkit verilog
make -f Makefile.vc707-u500devkit mcs
```
then write the bitstream to vc707.

The source and instruction of building linux of sd-boot image for vc707 with no-pcie.
The readme:
https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config
To build,
```
git clone https://github.com/mcd500/freedom-u-sdk.git
cd freedom-u-sdk
git submodule update --init --recursive --force
git checkout -b linux_u500vc707devkit_config remotes/origin/linux_u500vc707devkit_config
cd buildroot
git checkout -b sifive remotes/origin/sifive
cd ..
make -j4 BOARD=vc707devkit_nopci
```
This will write the image to SD card.
```
make DISK=/dev/whereisSDcard vc707-sd-write
```

Please let me know if anything does not work, so I could revise it before I create pull request.

mcd500, 2019-04-13:
This is the extracted log when booted on vc707:
```
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Core IP

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (12032512 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000bffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB [mem 0xbb1fb000-0xbf1fb000] (64MB) mapped at [        (ptrval)-     ]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 17 pages/cpu @        (ptrval) s28888 r8192 d32552 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 947396K/1046528K available (3042K kernel code, 214K rwdata, 1041K rodata, 11)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] riscv,plic0,c000000: mapped 6 interrupts to 4/8 handlers
[... 52 lines cut ...]
[    6.162939] libphy: Fixed MDIO Bus: probed
[    6.171570] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    6.179332] usbcore: registered new interface driver usb-storage
[    6.244939] mmc_spi spi0.0: SD/MMC host mmc0, no DMA, no WP, no poweroff, cd polling
[    6.257755] usbcore: registered new interface driver usbhid
[    6.262778] usbhid: USB HID core driver
[    6.272925] NET: Registered protocol family 17
[    6.294509] Loading compiled-in X.509 certificates
[    6.425714] mmc0: host does not support reading read-only switch, assuming write-enable
[    6.433375] mmc0: new SDHC card on SPI
[    6.445369] mmcblk0: mmc0:0000 SL08G 7.40 GiB 
[    6.679870] Freeing unused kernel memory: 11912K
[    6.683938] This architecture does not have kernel memory protection.
Starting logging: OK
Starting mdev...
sort: /sys/devices/platform/Fixed: No such file or directory
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... done.
Starting network...
Waiting for interface eth0 to appear............... timeout!
run-parts: /etc/network/if-pre-up.d/wait_iface: exit status 1

Welcome to Buildroot
buildroot login:
```

tmagik, 2019-04-15:
@erikdanie is there any way we can support both VC707 configs (with and without the PCI HMC expansion) from the same bitstream?

erikdanie, 2019-04-16:
I think this idea is good, and we need to make it easier to generate a no-pcie bitstream, but using sed in the makefile to comment out the line isn't the ideal way to do it. Instead, what should happen (and what I will do eventually when I get time, or if you want to try it on your own that would be awesome!) is that we should have 2 "shells" in VC707NewShell.scala. One class is the "VC707BasicShell", which includes all the peripherals and additions that are standard, and the other is "VC707PCIeShell", which extends VC707BasicShell, but then adds in the pcie. You can see an example of this in the master branch of fpga-shells in VCU118NewShell.scala. Then, in the makefile, depending on if the user passed in the "nopcie" argument, it would either do `MODEL = VC707BasicShell` or `MODEL := VC707PCIeShell`

Thanks for submitting this though! The makefile additions are helpful, and I'd merge those in separately if you were to put in a PR

mcd500, 2019-04-16:
Hi @tmagik and @erikdanie, thank you for the comments, I will revise them to using `MODEL = VC707BasicShell` or `MODEL := VC707PCIeShell` and make the pull request probably this weekend.

mcd500, 2019-04-19:
Hi @tmagik and @erikdanie, I pushed the changes to my repo.
I removed using `sed` and it will use `MODEL=` with selecting the build with/without PCIe board.
The VC707BaseShell extends VC707BaseShell for selecting which bitstream to build.
Is my changes matches your thoughts?
```sh
With PCIe-HMD board
$ make -f Makefile.vc707-u500devkit verilog
$ make -f Makefile.vc707-u500devkit mcs

For no PCIe-HMD board
$ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog
$ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs
```
The links are here. I tested and it looks working so might need to change the branch name before I make the pull request.
https://github.com/mcd500/freedom/tree/vc707-nopcie-switchable-not-fully-tested-builds-ok
https://github.com/mcd500/fpga-shells/commits/vc707-nopcie-switchable

mcd500, 2019-04-19:
The log when I tested the above changes.
```
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00047-g2751b6aad368-dirty (vc707@vc707-desktop) (gcc version 7.2.0 (GCC)) #1 SMP Fri Apr 19 11:01:18 JST 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9457664 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000bffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB [mem 0xbb1fb000-0xbf1fb000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28568 r0 d28776 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 950520K/1046528K available (2810K kernel code, 199K rwdata, 732K rodata, 9395K init, 777K bss, 96008K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] ------------[ cut here ]------------
[    0.000000] WARNING: CPU: 3 PID: 0 at /var/www/html/git/freedom-u-sdk/linux/drivers/irqchip/irq-riscv-intc.c:167 riscv_irq_enable+0x92/0x94
[... 71 lines cut ...]
[    5.174678] sifive_spi 64001000.spi: mapped; irq=2, cs=1
[    5.191002] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    5.198154] usbcore: registered new interface driver usb-storage
[    5.208174] usbcore: registered new interface driver usbhid
[    5.213192] usbhid: USB HID core driver
[    5.222408] NET: Registered protocol family 17
[    5.235770] Loading compiled-in X.509 certificates
[    5.528082] Freeing unused kernel memory: 9392K
[    5.532069] This architecture does not have kernel memory protection.
Starting logging: OK
Starting mdev...
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... done.
Starting network...
Waiting for interface eth0 to appear............... timeout!
run-parts: /etc/network/if-pre-up.d/wait_iface: exit status 1
Starting dropbear sshd: OK

Welcome to Buildroot
buildroot login: root
Password: 
# echo Great Working!
Great Working!
# 
```

tmagik, 2019-04-19:
> 
> Hi @tmagik and @erikdanie, I pushed the changes to my repo.
> I removed using sed and it will use MODEL= with selecting the build with/without PCIe board.
> The VC707BaseShell extends VC707BaseShell for selecting which bitstream to build.
> Is my changes matches your thoughts?
> 
> With PCIe-HMD board
> $ make -f Makefile.vc707-u500devkit verilog
> $ make -f Makefile.vc707-u500devkit mcs
> 
> For no PCIe-HMD board
> $ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog
> $ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs
> 
> The links are here. I tested and it looks working so might need to change the branch name before I make the pull request.
> https://github.com/mcd500/freedom/tree/vc707-nopcie-switchable-not-fully-tested-builds-ok
> https://github.com/mcd500/fpga-shells/commits/vc707-nopcie-switchable

This looks right to me. Does the build pass timing? I’ve had to slow some builds down to 100mhz clock to get it to pass timing without violations.

We also had a discussion today about moving from the FMC connector (which is quite expensive), and instead using an edge-card connector which I currently don’t have the part number or link for, and using the card-edge PCI, as it’s a lot simpler that way.

mcd500, 2019-04-19:
Hi @tmagik, yes the build passed timing, the default was `DevKitU500FPGADesign_WithDevKit50MHz` in the `Makefile.vc707-u500devkit`. Running on 50MHz was perfectly fine.
I also tried
```
$ make CONFIG=DevKitU500FPGADesign_WithDevKit100MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog
$ make CONFIG=DevKitU500FPGADesign_WithDevKit100MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs
```
and 
```
$ make CONFIG=DevKitU500FPGADesign_WithDevKit150MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog
$ make CONFIG=DevKitU500FPGADesign_WithDevKit150MHz MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs
```
The 100MHz was no problem. The 150Mhz did not pass the timing on the build but the generated bitstream is working OK so far :-)

wwMark, 2019-06-20:
> https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config

@mcd500 Hello, I have tried your vc707-nopci and linux_u500vc707devkit_config on my vc707 board, it could show the sifive logo on boot, but then it did nothing. Then I have tried vc707-nopcie-switchable-not-fully-tested-builds-ok and vc707-nopcie-switchable, but both of them had the same result as the vc707-nopci. Do you have a clue of what is going on?

mcd500, 2019-06-21:
Hi @wwMark , not really, it was working on my place at the office.
Did you tried the command I added bellow when you write to SD card?
```
sodo make DISK=/dev/whereisSDcard vc707-sd-write
```
since the bellow do not work for vc707.
```
sudo make DISK=/dev/sdX format-boot-loader
```
It is in the readme here.
https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config

wwMark, 2019-06-23:
> Hi @wwMark , not really, it was working on my place at the office.
> Did you tried the command I added bellow when you write to SD card?
>
> ```
> sodo make DISK=/dev/whereisSDcard vc707-sd-write
> ``` [...]

Hello @mcd500, I have tried with these commands, and they didn't work. Then my advisor tried them again, and everything succeeded now. We are trying to find out the reason. Thank you for your help.

s1104439112, 2019-07-01:
Hello @mcd500, thanks your work. Now, following your instructions I boot linux on VC707 without PCIe successfully. It's funny. But I am a newbie, I don't know how to run my program or accelerator, this Linux doesn't have some tools like gcc.So I ask for some advice about how to use this linux?
Thanks in advance.

mcd500, 2019-07-02:
Hi @s1104439112 , the current Linux boot image for vc707 is buildroot which is not able to install packages with apt or yum tools. I am pretty busy at the moment, so I am not sure when I could post it here, creating Debian rootfs for vc707 will be able to install gcc and etc for development.

jalezeta, 2019-09-05:
First of all thanks @mcd500 for your support, I can boot Linux without the external PCI clock. However, the eth0 port is not available, which I can appreciate it in your boot sequence as well (timeout message showing up). I guess this happens because we are not using the external PCIe NIC, which it would not work without the external PCIe clock generator. On top of that I do not see a reason why we need to use the external PCI clock generator since both clock frequencies in the range of 200MHz.

erikdanie, 2019-09-06:
Your above statement is correct. The reason the external generator is used is to more easily sync the PCIe clock and reset with the board, plus I believe the oscillator on the PCIe card is supposed to be better than the one on the board by default.

jalezeta, 2019-09-06:
Thank you @erikdanie for solving doubts!

mcd500, 2019-09-14:
Thanks, forgot to close this. The changes are merged at https://github.com/sifive/freedom/pull/118 and https://github.com/sifive/freedom/pull/133

closed 2019-09-14 by mcd500

minisparrow, 2019-11-18:
Hi @mcd500 , does vc707BaseShell support interface eth0 ? When i boot the fpga board and connect to the host cpu via ethnet wire, it appears as follows:
```
starting network...
Waiting for interface eth0 to appear............... timeout!
run-parts: /etc/network/if-pre-up.d/wait_iface: exit status 1
Starting dropbear sshd: OK
```
How should I enable the eth0 interface in the fpga board? Looking forward to your reply.Thanks very much.

krabo0om, 2019-11-18:
@minisparrow Hi, the FMC extension card offers a PCIe slot in which an ethernet card can be plugged. The on-board ethernet is not supported. You can have a look at the [Ariane Pulp Plattform](https://github.com/pulp-platform/ariane/), they appear to have at least some support for the VC707 on-board ethernet. Please let me know if you get it working without extension cards.

minisparrow, 2019-11-19:
> @minisparrow Hi, the FMC extension card offers a PCIe slot in which an ethernet card can be plugged. The on-board ethernet is not supported. You can have a look at the [Ariane Pulp [...]

thanks very much @krabo0om , I will have a look at that :)