sifive/freedom #119: Proposal of io address of vc707 to match HiFive Unleashed

mcd500, 2019-04-19:


mcd500, 2019-04-19:
This is just a proposal.
We are trying to run Keystone on vc707 with minimal changes in the source code of Keystone which runs perfectly on HiFive Unleashed.

These are the io address would make them matches with HiFive Unleashed.
Would be a problem for making pull request after testing them?

diff --git a/bootrom/sdboot/include/platform.h b/bootrom/sdboot/include/platform.h
index 73052bd..2b76f70 100644
--- a/bootrom/sdboot/include/platform.h
+++ b/bootrom/sdboot/include/platform.h
@@ -36,9 +36,9 @@
 #define CLINT_CTRL_SIZE _AC(0x10000,UL)
 #define DEBUG_CTRL_ADDR _AC(0x0,UL)
 #define DEBUG_CTRL_SIZE _AC(0x1000,UL)
-#define ERROR_MEM_ADDR _AC(0x3000,UL)
+#define ERROR_MEM_ADDR _AC(0x18000000,UL)
 #define ERROR_MEM_SIZE _AC(0x1000,UL)
-#define GPIO_CTRL_ADDR _AC(0x64002000,UL)
+#define GPIO_CTRL_ADDR _AC(0x10060000,UL)
 #define GPIO_CTRL_SIZE _AC(0x1000,UL)
 #define MASKROM_MEM_ADDR _AC(0x10000,UL)
 #define MASKROM_MEM_SIZE _AC(0x2000,UL)
@@ -46,11 +46,11 @@
 #define MEMORY_MEM_SIZE _AC(0x40000000,UL)
 #define PLIC_CTRL_ADDR _AC(0xc000000,UL)
 #define PLIC_CTRL_SIZE _AC(0x4000000,UL)
-#define SPI_CTRL_ADDR _AC(0x64001000,UL)
+#define SPI_CTRL_ADDR _AC(0x10040000,UL)
 #define SPI_CTRL_SIZE _AC(0x1000,UL)
 #define TEST_CTRL_ADDR _AC(0x4000,UL)
 #define TEST_CTRL_SIZE _AC(0x1000,UL)
-#define UART_CTRL_ADDR _AC(0x64000000,UL)
+#define UART_CTRL_ADDR _AC(0x10010000,UL)
 #define UART_CTRL_SIZE _AC(0x1000,UL)
 
 // IOF masks
diff --git a/bootrom/sdboot/linker/memory.lds b/bootrom/sdboot/linker/memory.lds
index 7f6cfc9..cf02376 100644
--- a/bootrom/sdboot/linker/memory.lds
+++ b/bootrom/sdboot/linker/memory.lds
@@ -1,14 +1,14 @@
 MEMORY
 {
   debug_ctrl (rwx) : ORIGIN = 0x0, LENGTH = 0x1000
-  error_mem (rw) : ORIGIN = 0x3000, LENGTH = 0x1000
+  error_mem (rw) : ORIGIN = 0x18000000, LENGTH = 0x1000
   test_ctrl (rw) : ORIGIN = 0x4000, LENGTH = 0x1000
   maskrom_mem (rx) : ORIGIN = 0x10000, LENGTH = 0x2000
   clint_ctrl (rw) : ORIGIN = 0x2000000, LENGTH = 0x10000
   plic_ctrl (rw) : ORIGIN = 0xc000000, LENGTH = 0x4000000
   axi_pcie_host_1_00_a_ctrl (rw) : ORIGIN = 0x50000000, LENGTH = 0x4000000
-  uart_ctrl (rw) : ORIGIN = 0x54000000, LENGTH = 0x1000
-  spi_ctrl (rw) : ORIGIN = 0x54001000, LENGTH = 0x1000
-  gpio_ctrl (rw) : ORIGIN = 0x54002000, LENGTH = 0x1000
+  uart_ctrl (rw) : ORIGIN = 0x10010000, LENGTH = 0x1000
+  spi_ctrl (rw) : ORIGIN = 0x10040000, LENGTH = 0x1000
+  gpio_ctrl (rw) : ORIGIN = 0x10060000, LENGTH = 0x1000
   memory_mem (rwx) : ORIGIN = 0x80000000, LENGTH = 0x40000000
 }
diff --git a/fpga-shells b/fpga-shells
--- a/fpga-shells
+++ b/fpga-shells
@@ -1 +1 @@
-Subproject commit 9eb56625ba252e6abbb7ded5158331b7da7ea39a
+Subproject commit 9eb56625ba252e6abbb7ded5158331b7da7ea39a-dirty
diff --git a/src/main/scala/unleashed/DevKitConfigs.scala b/src/main/scala/unleashed/DevKitConfigs.scala
index 207c90a..3e4b4ff 100644
--- a/src/main/scala/unleashed/DevKitConfigs.scala
+++ b/src/main/scala/unleashed/DevKitConfigs.scala
@@ -24,11 +24,11 @@ class FreedomU500Config extends Config(
 // Freedom U500 Dev Kit Peripherals
 class U500DevKitPeripherals extends Config((site, here, up) => {
   case PeripheryUARTKey => List(
-    UARTParams(address = BigInt(0x64000000L)))
+    UARTParams(address = BigInt(0x10010000L)))
   case PeripherySPIKey => List(
-    SPIParams(rAddress = BigInt(0x64001000L)))
+    SPIParams(rAddress = BigInt(0x10040000L)))
   case PeripheryGPIOKey => List(
-    GPIOParams(address = BigInt(0x64002000L), width = 4))
+    GPIOParams(address = BigInt(0x10060000L), width = 4))
   case PeripheryMaskROMKey => List(
     MaskROMParams(address = 0x10000, name = "BootROM"))
 })
@@ -40,7 +40,7 @@ class U500DevKitConfig extends Config(
   new FreedomU500Config().alter((site,here,up) => {
     case SystemBusKey => up(SystemBusKey).copy(
       errorDevice = Some(DevNullParams(
-        Seq(AddressSet(0x3000, 0xfff)),
+        Seq(AddressSet(0x18000000, 0xfff)),
         maxAtomic=site(XLen)/8,
         maxTransfer=128,
         region = RegionType.TRACKED)))

mcd500, 2019-04-19:
This is the log of Keystone running on vc707 after changing the io addresses.
Not perfectly but starting to working.
```
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT
bbl loader
secure boot
create a random seed
copy TEST device key
measure SM
generate sm_key_seed
generate public keys
endose and sign
clean up and go

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Core IP
miiininnintititiaialaliliziziziningng g s smsm
tinit

d iaialinilniiztziienadgl  isszmme
 i 
   tisma
zedzneitid salim
    sm
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00060-g65e929792fb9-dirty (kkojima@kkojima-ThinkPad-P70) (gcc version 8.2.0 (GCC)) #45 SMP Mon Apr 19
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (12032512 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000bffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB [mem 0xbb1fb000-0xbf1fb000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 16 pages/cpu @        (ptrval) s28632 r8192 d28712 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[... 188 lines cut ...]
Enclave said value: 20
testing attestation
[  258.913162] INFO: rcu_sched self-detected stall on CPU
[  258.917772]  2-....: (2099 ticks this GP) idle=2a6/140000000000001/0 softirq=887/887 fqs=12 
[  258.926102]   (t=2100 jiffies g=126 c=125 q=2)
[  258.930550] Task dump for CPU 2:
[  258.933764] test-runner.ris R  running task        0   132    124 0x00000008
[  258.940794] Call Trace:
[  258.943348] [<        (ptrval)>] walk_stackframe+0x0/0xa2
[  258.948687] [<        (ptrval)>] show_stack+0x26/0x34
[  258.953743] [<        (ptrval)>] sched_show_task+0xa8/0x108
[  258.959270] [<        (ptrval)>] dump_cpu_task+0x3a/0x48
[  258.964575] [<        (ptrval)>] rcu_dump_cpu_stacks+0x7a/0xb8
[  258.970392] [<        (ptrval)>] rcu_check_callbacks+0x52c/0x644
[  258.976406] [<        (ptrval)>] update_process_times+0x1e/0x4c
[  258.982318] [<        (ptrval)>] tick_periodic+0x40/0xac
[  258.987606] [<        (ptrval)>] tick_handle_periodic+0x56/0x5c
[  258.993536] [<        (ptrval)>] riscv_timer_interrupt+0x26/0x32
[  258.999541] [<        (ptrval)>] riscv_intc_irq+0x58/0xee
[  259.004912] [<        (ptrval)>] ret_from_syscall+0xa/0xe
[  259.010418] [<        (ptrval)>] keystone_finalize_enclave+0x14/0xe4 [keystone_driver]
Attestation report SIGNATURE is valid
# 

```

tmagik, 2019-04-20:
Have a look at https://github.com/tmagik/freedom/blob/bootrom_hacks/bootrom/uboot/Makefile

mcd500, 2019-04-20:
Your link is going to support different dts for every different boards for the kernel. e.g. Unleashed, vc707, vcu118. It is great. When the vc707 or Unleashed have extention boards or peripherals added, then probably using device tree overlay is the genuine dts way. which we could work on it some time in the future. Thanks for the link.