sifive/freedom #120: Build error when I pump up submodule to fpga-shells:master

mcd500, 2019-04-23:
I discovered that when I was making the pull request bellow,
https://github.com/sifive/freedom/pull/118
when I tried updating the submodule fpga-shells commit 9eb56625ba252e6abbb7ded5158331b7da7ea39a to the latest in master,
it generated the error message bellow.
```
[error] /home/riscv/projects/freedom/src/main/scala/unleashed/IOFPGADesign.scala:65:63: not enough arguments for method apply: (baseAddress: BigInt, wrangler: sifive.fpgashells.clocks.ClockAdapterNode, corePLL: sifive.fpgashells.clocks.PLLNode)(implicit p: freechips.rocketchip.config.Parameters)sifive.fpgashells.shell.DDROverlayParams in object DDROverlayParams.
[error] Unspecified value parameter corePLL.
[error]   val ddr = p(DDROverlayKey).headOption.map(_(DDROverlayParams(0x3000000000L, wrangler.node)))
[error]                                                               ^
[error] /home/riscv/projects/freedom/src/main/scala/unleashed/IOFPGADesign.scala:118:19: value := is not a member of Nothing
[error]   Expression does not convert to assignment because receiver is not assignable.
[error]   ddr.foreach { _ := sbar.node }
[error]                   ^
[error] /home/riscv/projects/freedom/src/main/scala/unleashed/vera/iofpga/FPGAChip.scala:143:64: type mismatch;
[error]  found   : sifive.fpgashells.clocks.PLLParameters
[error]  required: sifive.fpgashells.ip.microsemi.polarfireccc.PolarFireCCCParameters
[error]     val chiplink_rx_pll = Module(new PolarFireCCC(PLLParameters(
[error]                                                                ^
[error] three errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 33 s, completed Apr 19, 2019, 2:04:51 PM
common.mk:48: recipe for target '/home/riscv/projects/freedom/builds/vc707-u500devkit/sifive.freedom.unleashed.DevKitU500FPGADesign_WithDevKit100MHz.fir' failed
make: *** [/home/riscv/projects/freedom/builds/vc707-u500devkit/sifive.freedom.unleashed.DevKitU500FPGADesign_WithDevKit100MHz.fir] Error 1

```
It seems that when `corePLL: PLLNode` was added to `case class DDROverlayParams` in `src/main/scala/shell/DDROverlay.scala` it became out of sync with other places.

Do you mind suggesting me what would be a good way to solve them?

mcd500, 2019-04-23:
So I did not bump up the submodule `fpga-shells` in my pull request  https://github.com/sifive/freedom/pull/118

erikdanie, 2019-04-23:
Yeah, I am actively working on getting the submodules bumped, but due to some neglect, a lot of technical debt has built up. I should get it done soon

mcd500, 2019-04-24:
I am here to help you, please let me know if I could help anything, since everybody looks very busy.