sifive/freedom #122: Free vivado version sufficient ?

bagyavinoth, 2019-04-29:
Hello,
I see that below vivado version is free of cost with limited functionalities.
Vivado Design Suite HL WebPACK

Is this sufficient to work with freedom platforms or do we need the paid version ?

Regards,
Vinoth

erikdanie, 2019-04-29:
Yes, the free version should work fine to generate the freedom builds

bagyavinoth, 2019-05-18:
Thanks for the response @erikdanie .
And i hope it's applicable to both the below platforms.
Freedom E300 Arty FPGA Dev Kit 
Freedom U500 VC707 FPGA Dev Kit
Right ?

Regards,
Vinoth

richardxia, 2019-05-18:
Unfortunately, it only applies to the Arty boards. The VC707 requires a paid version of Vivado; however, the purchase of a VC707 board will come with a node-locked license of Vivado that can be used to synthesize bitstreams for the board.

bagyavinoth, 2019-05-18:
Ho ok. Thanks @richardxia.

closed 2019-05-18 by bagyavinoth