sifive/freedom #126: Possible to synthesize E300 and U500 on the Polarfire Evaluation Board

carlosedp, 2019-07-05:
Is it possible to synthesize these cores on the [Polarfire Evaluation kit](https://www.microsemi.com/existing-parts/parts/138273) (300k le) and if yes, what changes are required?

erikdanie, 2019-07-08:
Currently, this board is not supported. It would need to have a shell written for it (see fpga-shells/src/main/scala/microsemi/VeraShell.scala), but that shell is currently targeted for iofpga and lacks many peripherals that the evaluation kit would have. You could also try to make verilog and use the Libero tools to map to that board, but I haven't tried that before with the Libero tools

carlosedp, 2019-07-08:
I see this shell too in the same dir: https://github.com/sifive/fpga-shells/blob/9eb56625ba252e6abbb7ded5158331b7da7ea39a/src/main/scala/shell/microsemi/PolarFireEvalKitShell.scala

Is that it?

erikdanie, 2019-07-09:
That actually may be, that is from before I started and the person who would know about it is on vacation. You may be able to make use of that via a makefile, but this is currently not something we support. If you are savvy with Chisel and want to give it a try, feel free to, but you will likely run into some problems