sifive/freedom #127: Error: Target not examined yet

nf9, 2019-07-24:
Hello,
I have successfully build and uploaded Freedom E300 on my Arty Rev. C board using Vivado 2018.3. OpenOCD can detect the core but when I send telnet command to it, it returns '''Error: Target not examined yet'''

''' 
nf@ubuntu:~/opt/FreedomStudio/SiFive/riscv-openocd/riscv-openocd-0.10.0-2019.05.1/bin$ ./openocd -f /home/nf/ws/RISCV/freedom-e310-arty-hello/bsp/openocd.cfg 
Open On-Chip Debugger 0.10.0+dev (SiFive OpenOCD 0.10.0-2019.05.1)
Licensed under GNU GPL v2
For bug reports:
	https://github.com/sifive/freedom-tools/issues
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Info : Listening on port 3333 for gdb connections
Info : accepting 'telnet' connection on tcp/4444
Error: Target not examined yet
'''

Any ideas?

erikdanie, 2019-08-06:
Hi,

For issues like this, I recommend posting on the [SiFive Forums](https://forums.sifive.com)

How are you sending a telnet command to it? And does that Error line only pop up when you try to send a command?

nf9, 2019-08-08:
I open another terminal and run telnet to the target OpenOCD. That is correct. When I try to send a command (init, reset, flash, etc) it returns `Error: Target not examined yet`

erikdanie, 2019-08-08:
Can you add the -d flag to the terminal where you are running openocd and see the results that get spit out when it errors out?

nf9, 2019-08-12:
here is the output with -d option
'''
nf@ubuntu:~/opt/FreedomStudio/SiFive/riscv-openocd/riscv-openocd-0.10.0-2019.05.1/bin$ ./openocd -f /home/nf/ws/RISCV/freedom-e310-arty-hello/bsp/openocd.cfg -d
Open On-Chip Debugger 0.10.0+dev (SiFive OpenOCD 0.10.0-2019.05.1)
Licensed under GNU GPL v2
For bug reports:
	https://github.com/sifive/freedom-tools/issues
User : 13 1 command.c:544 command_print(): debug_level: 3
Debug: 14 1 options.c:184 add_default_dirs(): bindir=/var/lib/builds/sifive-tools/freedom-tools-qemu-src/obj/x86_64-linux-centos6/install/riscv-openocd-0.10.0-2019.05.1-x86_64-linux-centos6/bin
Debug: 15 1 options.c:185 add_default_dirs(): pkgdatadir=/var/lib/builds/sifive-tools/freedom-tools-qemu-src/obj/x86_64-linux-centos6/install/riscv-openocd-0.10.0-2019.05.1-x86_64-linux-centos6/share/openocd
Debug: 16 1 options.c:186 add_default_dirs(): exepath=/home/nf/opt/FreedomStudio/SiFive/riscv-openocd/riscv-openocd-0.10.0-2019.05.1/bin
Debug: 17 1 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 18 1 configuration.c:42 add_script_search_dir(): adding /home/nf/.openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/nf/opt/FreedomStudio/SiFive/riscv-openocd/riscv-openocd-0.10.0-2019.05.1/bin/../share/openocd/site
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/nf/opt/FreedomStudio/SiFive/riscv-openocd/riscv-openocd-0.10.0-2019.05.1/bin/../share/openocd/scripts
Debug: 21 1 configuration.c:97 find_file(): found /home/nf/ws/RISCV/freedom-e310-arty-hello/bsp/openocd.cfg
Debug: 22 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 10000
Debug: 23 1 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 10000
Debug: 25 1 core.c:1645 jtag_config_khz(): handle jtag khz
Debug: 26 1 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 27 1 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 28 1 command.c:544 command_print(): adapter speed: 10000 kHz
Debug: 29 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface ftdi
Debug: 30 1 command.c:143 script_debug(): command - interface ocd_interface ftdi
Debug: 32 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_device_desc'...
Debug: 33 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_serial'...
Debug: 34 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_channel'...
Debug: 35 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_init'...
Debug: 36 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_signal'...
Debug: 37 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_set_signal'...
Debug: 38 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_get_signal'...
Debug: 39 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_vid_pid'...
Debug: 40 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_tdo_sample_edge'...
Debug: 41 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_oscan1_mode'...
Debug: 42 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_device_desc Olimex OpenOCD JTAG ARM-USB-OCD-H
Debug: 43 1 command.c:143 script_debug(): command - ftdi_device_desc ocd_ftdi_device_desc Olimex OpenOCD JTAG ARM-USB-OCD-H
Debug: 45 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_vid_pid 0x15ba 0x002b
Debug: 46 1 command.c:143 script_debug(): command - ftdi_vid_pid ocd_ftdi_vid_pid 0x15ba 0x002b
Debug: 48 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_init 0x0908 0x0b1b
Debug: 49 1 command.c:143 script_debug(): command - ftdi_layout_init ocd_ftdi_layout_init 0x0908 0x0b1b
Debug: 51 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_signal nSRST -oe 0x0200
Debug: 52 2 command.c:143 script_debug(): command - ftdi_layout_signal ocd_ftdi_layout_signal nSRST -oe 0x0200
Debug: 54 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_signal nTRST -data 0x0100
Debug: 55 2 command.c:143 script_debug(): command - ftdi_layout_signal ocd_ftdi_layout_signal nTRST -data 0x0100
Debug: 57 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_signal LED -data 0x0800
Debug: 58 2 command.c:143 script_debug(): command - ftdi_layout_signal ocd_ftdi_layout_signal LED -data 0x0800
Debug: 60 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 61 2 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 62 2 transport.c:286 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 63 2 command.c:364 register_command_handler(): registering 'ocd_jtag_flush_queue_sleep'...
Debug: 64 2 command.c:364 register_command_handler(): registering 'ocd_jtag_rclk'...
Debug: 65 2 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 66 2 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_assert_width'...
Debug: 67 2 command.c:364 register_command_handler(): registering 'ocd_scan_chain'...
Debug: 68 2 command.c:364 register_command_handler(): registering 'ocd_jtag_reset'...
[... cut ...]
Debug: 734 3097 tcl.c:497 handle_nand_init_command(): Initializing NAND devices...
Debug: 735 3097 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_pld init
Debug: 736 3097 command.c:143 script_debug(): command - ocd_pld ocd_pld init
Debug: 738 3097 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 739 3097 gdb_server.c:3425 gdb_target_start(): starting gdb server for riscv.cpu.0 on 3333
Info : 740 3097 server.c:311 add_service(): Listening on port 3333 for gdb connections
'''