sifive/freedom #129: Connect a signal from Rocket core class to FPGA chip class to test on FPGA board?

phuong2403, 2019-08-09:


erikdanie, 2019-08-09:
Hi! You might want to try posting on the [SiFive Forums](https://forums.sifive.com/). You'll find more people there who are working on similar things. 

Without being able to see your code, the pertinent part of this error message appears to be:

> [error] Caused by: chisel3.internal.ChiselException: Error: Called Module() twice without instantiating a Module. See @[RocketTile.scala 44:22]

and that seems to be occurring indicated by this line:

>sifive.freedom.everywhere.e300artydevkit.E300ArtyDevKitFPGAChip.$anonfun$new$5(FPGAChip.scala:41)

(line 41 of FPGAChip)

phuong2403, 2019-08-12:
Thank you for your reply @erikdanie.
The thing is I tried to call the signal Trigger, which is in Rocket class, at the RocketTile class by the inheriting: val core = Module(new Rocket()). I did it because I saw that by calling this signal in this class, in the next wrap (SubSystem), the RocketTile class is inherited, so I can put the signal Trigger out to Top step by step.

However, in another class name RocketTileModuleImp, the Rocket class has already called: val core = Module(new Rocket()(outer.p)).

So could you give me some advice?

austinharris, 2019-08-12:
You might want to look into this feature of chisel: https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/experimental/BoringUtils.scala

phuong2403, 2019-08-12:
Woah, I have just looked around it. It seems to be useful to me.
Thank you so much @austinharris.
I will try with it now