sifive/freedom #131: Need Assistant for Flowchart of U500-VC707 to Run Liunx

minmossi, 2019-08-27:
Currently I have VC707 board only, without any FMC-PCIe board and PCIe switch/device.
I'm trying to program VC707 with U500 .mcs file and boot Linux, but I'm not sure I'm doing right.

1. Generate .mcs/.prm for Programming VC707
```
$ git clone https://github.com/sifive/freedom.git
$ cd freedom
$ git submodule update --init --recursive
$ export RISCV=/where/riscv_tools/located
$ source /where/Xilinx/Vivado/located/settings64.sh
$ make -f Makefile.vc707-u500devkit mcs
```

2. Generate SD Boot Image
```
$ git clone https://github.com/sifive/freedom-u-sdk.git
$ cd freedom-u-sdk
$ git submodule update --recursive --init
$ make
$ sudo make DISK=/dev/sd_card_location format-boot-loader
```

3. Programming FPGA with .mcs/.prm
```
1. Plug USB Type-A to mini-USB cable to JTAG port on VC707's backplate
2. Setting SW11 to 11010 on VC707
3. Power-up VC707
4. Open Vivado 2016.4
5. Open Hardware Manager
6. Create Configuration Memory Device(mt28gu01gaaxle-bpi-x16)
7. select .mcs/.prm
8. State of non-config mem I/O pins: Pull-none
9. Address Range: Configuration File Only
10. RS Pins: 25:24
11. Erase: Yes
12. Blank Check: No
13. Program: Yes
14. Verify: Yes
15. Verify Checksum: No
```

4. Prepare to Boot Linux
```
1. Insert programmed 16GB SDHC card to VC707 board
2. Plug USB Type-A to micro-USB cable to UART port on VC707's backplate
3. $ sudo minicom /dev/ttyUSB0
```

But I don't have anything printed on minicom screen, also everytime I turn VC707 off the FPGA returns to unprogrammed state or the configuration memory gets deleted.

Any help I would be thankful for.

minmossi, 2019-08-28:
> The vc707 'rom' bootloader needs to be updated to support GPT partitions before the master branch freedom-u-sdk branch will work.

Then what should I do? linux_u500vc707devkit_config branch of freedom-u-sdk is broken due to broken buildroot/m4. Should I just patch [the resoultion](https://github.com/buildroot/buildroot/commit/c48f8a64626c60bd1b46804b7cf1a699ff53cdf3) to linux_u500vc707devkit_config branch?

mcd500, 2019-08-28:
Hi @minmossi.  I have been doing the same as making the vc707 boot linux with and without FMC-PCIe board.
I am middle up upstreaming with the help from @erikdanie, in these discussion. 
https://github.com/sifive/freedom/issues/116
https://github.com/sifive/fpga-shells/pull/64
https://github.com/sifive/freedom/pull/118
In the mean time you could try my repo.
First download riscv64-elf-gcc-8.2 from [here](https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-8.2.0-2019.05.3-x86_64-linux-ubuntu14.tar.gz)
and untar it some place in your home directory.
1. Generate .mcs/.prm for Programming VC707
```
$ git clone https://github.com/mcd500/freedom.git
$ cd freedom
$ git checkout vc707-nopcie-switchable-mcd500
$ git submodule update --init --recursive
$ export RISCV=/home/where_is_gcc/riscv64-elf-gcc-8.2
$ export PATH=/where_is_Vivado/2016.4/bin/:/home/where_is_gcc/riscv64-elf-gcc-8.2/bin/:${PATH}
$ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit verilog -j `nproc`
$ make MODEL=VC707BaseShell -f Makefile.vc707-u500devkit mcs -j `nproc`
```
And then write the bitstream to vc707.

2.  Generate SD Boot Image
```
$ git clone https://github.com/mcd500/freedom.git
$ cd freedom-u-sdk
$ git checkout linux_u500vc707devkit_config
$ git submodule update --init --recursive
$ make BOARD=vc707devkit_nopci -j `nproc`
$ sudo make DISK=/dev/sd_card_location  vc707-sd-write
```
Then turn on vc707 with the sd card and see the serial console.
It should show up boot message.
Please let me know if anything does not work.

minmossi, 2019-08-30:
@mcd500 Can you patch [this](https://github.com/buildroot/buildroot/commit/c48f8a64626c60bd1b46804b7cf1a699ff53cdf3) to your freedom-u-sdk branch? Using glibc 2.29 on Ubuntu 19.04, I can't build this due to following error:
```
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o freadahead.o freadahead.c
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fseek.o fseek.c
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fseeko.o fseeko.c
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o isnanl.o isnanl.c
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o printf-args.o printf-args.c
/usr/bin/gcc  -I.   -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o printf-parse.o printf-parse.c
freadahead.c: In function 'freadahead':
freadahead.c:91:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   ^~~~~
make[6]: *** [Makefile:1842: freadahead.o] Error 1
make[6]: *** Waiting for unfinished jobs....
fseeko.c: In function 'rpl_fseeko':
fseeko.c:109:4: error: #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
   #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
    ^~~~~
make[6]: *** [Makefile:1842: fseeko.o] Error 1
make[5]: *** [Makefile:1602: all] Error 2
make[4]: *** [Makefile:1506: all-recursive] Error 1
make[3]: *** [Makefile:1461: all] Error 2
make[2]: *** [package/pkg-generic.mk:198: /home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17/.stamp_built] Error 2
make[1]: *** [Makefile:36: _all] Error 2
make[1]: Leaving directory '/home/jihong/Sources/freedom-u-sdk/buildroot'
make: *** [Makefile:96: /home/jihong/Sources/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

minmossi, 2019-09-02:
I was able to fully synthesizing U500 for VC707 and boot Linux.

If you don't have HMC-PCIe daughter board, like me, you can use mcd500's repository following above, thanks Tsukamoto!

Those who using glibc > 2,28(like the ones who using latest Ubuntu), you need to patch [this](https://github.com/buildroot/buildroot/commit/c48f8a64626c60bd1b46804b7cf1a699ff53cdf3) to work/build_initramfs/build/host-m4-1.4.17 in freedom-u-sdk local repository.

Closing this issue now.

closed 2019-09-02 by minmossi

mcd500, 2019-09-02:
Hi @minmossi, you are most welcome!
I have a lot to upstream. I also made freedom-u-sdk on vc707 to read gpt partition of SD card and then making debian rootfs to boot from SD card on vc707. Need to cleanup and finish upstreaming.

manox, 2020-02-27:
Hi @mcd500 , I am very interested in reading GPT partitions and booting debian on VCU118. Did you publish your work for VC707 in the meantime?

mcd500, 2020-02-29:
Hi @manox, seems like I completely forgotten to upstream to here. But I upstream the sd card partitioning part at here. Reading the GPT partitions and booting on the vcu118 is probably the same.
https://github.com/keystone-enclave/keystone/blob/master/KEYSTONE-FPGA.md#prepare-keystone-bootable-sd-card

jctullos, 2020-04-06:
Hey @mcd500, I've been following the instructions here, and in other VC707 issue threads but something isn't working. The best I've got was UART output that stalled on CMD58. I suspect it's something for the bbl.bin and sd card partition, that but I can't seem to find a solid answer.  I've followed everything with no PCIE FMC module, and I'm using Vivado 2018.3. I've tried downgrading to 2016.4 in case that's causing the SDBoot to fail.

For the image for the SD Card, I'm following your fork at: https://github.com/mcd500/freedom-u-sdk and the linux_u500vc707devkit_config branch

I've tried 2 different SD cards, using the make -j4 BOARD=vc707devkit_nopci as per instructions. Then formatting the card with format-boot-loader.  

Thank you for all the help and all the instructions you provide for the VC707, they are extremely helpful!